espressif/esp-idf #1196: Can ESP32 parallell processes the Bluetooth and Wi-Fi?

vishnunaik, 2017-10-31:
Hi, 
I am having one question, as ESP32 consists of both Bluetooth and Wi-Fi can it be parallelly process the work of both Bluetooth and Wi-Fi. 
Consider a example: 
Initially ESP32 acts as gattclient to receive the data from the external gattserver, as soon as the data reached it should send the same data to the cloud through a Wi-Fi network .
Is this possible with the ESP32 module? If it so which program is suitable to do this?
Is there any program in esp-idf/examples please share with me.
Thank you.

akshar001, 2017-10-31:
Yes it can ....

https://github.com/ibm-watson-iot/device-arduino/blob/master/samples/ESP32BleGateway/ESP32BleGateway.ino

this  is an example you can do it like that..

vishnunaik, 2017-10-31:
Dear @akshar001 did you executed this code in arduino? If it so can you help us with the setup because we are new to the arduino environment  or you can share us the video link of this program setup and testing.
Thank you.

closed 2018-02-24 by FayeY

vishnunaik, 2018-05-31:
@akshar001  No, it is not working. we have worked on it for two months but we couldn't able to get any output.

haarislabs, 2018-07-09:
Hey @vishnunaik ,
Try using this example https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_gatts_mqtt_gateway

There's video on it as well. Check out his youtube channel

Gamecock, 2018-08-03:
Any reason this would not work as GATT Server?