espressif/esp-idf #1198: Boot Loop BLE+Wifi

omerfatih, 2017-10-31:
Hi,

I am developing a GATEWAY code for my project using WiFi and BLE simultaneously. However when I initialized both device, board enters to a bootloop after initialize app-core. The output trace/debug is like below. I can't find any resource how to use WiFi and BLE together. I am putting the code at the end of my message. Please help me. 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:5576
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:12476
entry 0x40078f74
W (74) rtc_clk: Bogus XTAL frequency: 0 MHz
I (35) boot: ESP-IDF v3.0-dev-806-gde750e99 2nd stage bootloader
I (35) boot: compile time 16:04:32
I (35) boot: Enabling RNG early entropy source...
I (41) boot: SPI Speed      : 40MHz
I (46) boot: SPI Mode       : DIO
I (50) boot: SPI Flash Size : 4MB
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x329b0 (207
280) map
I (172) esp_image: segment 1: paddr=0x000429d8 vaddr=0x3ffc0000 size=0x030a4 ( 1
2452) load
I (177) esp_image: segment 2: paddr=0x00045a84 vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _iram_start at C:/msys32/home/Kayten/esp/esp-idf/components/freertos
/xtensa_vectors.S:1675

I (179) esp_image: segment 3: paddr=0x00045e8c vaddr=0x40080400 size=0x0a184 ( 4
1348) load
I (205) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x9b584 (63
6292) map
0x400d0018: _stext at ??:?

I (425) esp_image: segment 5: paddr=0x000eb5a4 vaddr=0x4008a584 size=0x08548 ( 3
4120) load
0x4008a584: correct_rfpll_offset at ??:?

I (439) esp_image: segment 6: paddr=0x000f3af4 vaddr=0x400c0000 size=0x00000 (
   0) load
I (450) boot: Loaded app from partition at offset 0x10000
I (451) boot: Disabling RNG early entropy source...
I (451) cpu_start: Pro cpu up.
I (455) cpu_start: Starting app cpu, entry point is 0x40080ec8
0x40080ec8: call_start_cpu1 at C:/msys32/home/Kayten/esp/esp-idf/components/esp3
2/cpu_start.c:219

I (444) cpu_start: App cpu up.
I (466) heap_init: Initializing. RAM available for dynamic allocation:
I (472) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (478) heap_init: At 3FFDB590 len 00004A70 (18 KiB): DRAM
I (484) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (491) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (497) heap_init: At 40092ACC len 0000D534 (53 KiB): IRAM
I (503) cpu_start: Pro cpu start user code
I (186) cpu_start: Starting scheduler on PRO CPU.
abort() was called at PC 0x40080e8d on core 0
0x40080e8d: start_cpu0_default at C:/msys32/home/Kayten/esp/esp-idf/components/e
sp32/cpu_start.c:350 (discriminator 1)

Backtrace: 0x400898ec:0x3ffe3bb0 0x400899eb:0x3ffe3bd0 0x40080e8d:0x3ffe3bf0 0x4
0081081:0x3ffe3c20 0x40078c32:0x3ffe3c40 0x40078ce5:0x3ffe3c70 0x40078f4e:0x3ffe
3cb0 0x40079093:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x400898ec: invoke_abort at C:/msys32/home/Kayten/esp/esp-idf/components/esp32/p
anic.c:553

0x400899eb: abort at C:/msys32/home/Kayten/esp/esp-idf/components/esp32/panic.c:
553

0x40080e8d: start_cpu0_default at C:/msys32/home/Kayten/esp/esp-idf/components/e
sp32/cpu_start.c:350 (discriminator 1)

0x40081081: call_start_cpu0 at C:/msys32/home/Kayten/esp/esp-idf/components/esp3
2/cpu_start.c:207

Rebooting...

i am working this code: https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_gatts_mqtt_gateway

vishnunaik, 2017-11-01:
@omerfatih  try this code  [link:](https://github.com/ibm-watson-iot/device-arduino/blob/master/samples/ESP32BleGateway/ESP32BleGateway.ino)

omerfatih, 2017-11-01:
i am working ion esp-idf 3 , are you have code for esp idf ?

chegewara, 2017-11-05:
This is the clue why you are having boot loop
```
I (472) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (478) heap_init: At 3FFDB590 len 00004A70 (18 KiB): DRAM
I (484) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (491) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (497) heap_init: At 40092ACC len 0000D534 (53 KiB): IRAM
```
With 18kB DRAM on program start, before any initializations you can do nothing. You have to do some serious optimizations to make it work.

closed 2018-02-24 by FayeY