espressif/esp-idf #1200: BT: How to respond to long/reliable write requests?

rojer, 2017-10-31:
ESP32 is a GATT server. Client (actually, another ESP32) is trying to perform a Long Write request.
It sends a Prepare Write request (`write.is_prep = 1`) for the first MTU-3 bytes and, according to the BT spec, the server should respond with Prepare Write Response replaying the same data.
however, `esp_ble_gatts_send_response` does not take the value argument and seemingly does not do what is necessary either, since client gives up and sends an Execute Write with `exec_write.exec_write_flag = 0`, aka `ESP_GATT_PREP_WRITE_CANCEL`.

so the question is: how to respond to prepare write requests properly?

negativekelvin, 2017-10-31:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server/main/gatts_demo.c#L274

rojer, 2017-10-31:
oh, good. thanks!

closed 2017-10-31 by rojer