espressif/esp-idf #1201: lwip_select() stopped working for sockets

SStorm, 2017-10-31:
This seems  to have happened following commit e5ae0f5927048e6ac98d96f61b92c22d95f80a21 in the lwip component. The socket FDs returned from lwip_socket() were previously starting at 0, however now an offset is applied and they start at 4096, but lwip_select can work with a maximum fd of 64. 

Was this intentional? If so, is there some way to continue using lwip_select() ?

Thanks!

negativekelvin, 2017-10-31:
https://github.com/espressif/esp-idf/issues/1141

projectgus, 2017-10-31:
@SStorm As per the issue @negativekelvin linked, there was a bug where newlib FD_SET macros were being included instead of  LWIP ones. The LWIP ones should apply the offset correctly.

If you're still seeing the bug with current IDF master, can you show us the relevant code & headers which are being included?

SStorm, 2017-11-01:
The headers were in fact correct, however the semantics of how one is supposed to use select() have seemingly changed. i.e., normally you'd be doing:

```
ready_fd_count = select(FD_SETSIZE, &read_fd_set, NULL, NULL, &timeout)) < 0
```

However, lwip_select loops over the socket descriptors in an interesting way:
```
for (i = LWIP_SOCKET_OFFSET; i < maxfdp1; i++) {
```
where maxfdp1 is the parameter FD_SETSIZE. Basically, you need to call select() with FD_SETSIZE + LWIP_SOCKET_OFFSET, which kind of sucks and breaks POSIX compatibility (which is the whole point).

After that, you need to remember to iterate in the correct way as well:
```
for (int i = 0; i < FD_SETSIZE + LWIP_SOCKET_OFFSET; i++) {
    if (FD_ISSET (i, &read_fd_set)) { 
...
```

Am I missing something? Is this a bug or intended behaviour?

karawin, 2017-11-01:
The first parameter of the  call to select is not FD_SETSIZE, but the highest socket number+1.
Doing this, the new lwip is working very well.

SStorm, 2017-11-01:
Thanks, yes, figured it out too. Will close issue.

closed 2017-11-01 by SStorm