espressif/esp-idf #1203: i2s_adc_dac example i2s read value always zero

daiyinger, 2017-10-31:
anyone find this?

phonec, 2017-11-04:
I confirm the same thing, I tried.
Maybe there is not enough initialization of ADС? But I tried the standard settings - did not help

======
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
======
Sound recording 5%
======
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
........ ETC

phonec, 2017-11-22:
In this example, I replaced ADC initialization function and initialized the SAR ADC2 for scanning, I see that the DMA is being filled correctly, everything works only for SAR ADC2. Other function for initializing the SAR ADC1 does not work, DMA filled by nulls (or nothing filled) what's the reason? (my chip revision is rev0), 

/* WORKED FUNCTION */
void sar2_scan_init(void)
{
    /* SAR ADC2, 0 - 15 means pattern table length of 1 - 16. (R/W)
     * SET_PERI_REG_BITS(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR2_PATT_LEN, 0, SYSCON_SARADC_SAR2_PATT_LEN_S);
     * Test pattern configuration byte 0xAD:
     * --[7:4] channel_sel: 10-->en_test
     * --[3:2] bit_width  : 3-->12bit
     * --[1:0] atten      : 1-->3dB attenuation */
    WRITE_PERI_REG(SYSCON_SARADC_SAR2_PATT_TAB1_REG, 0x0E1E2E3E);
    WRITE_PERI_REG(SYSCON_SARADC_SAR2_PATT_TAB2_REG, 0x4E5E6E7E);
    WRITE_PERI_REG(SYSCON_SARADC_SAR2_PATT_TAB3_REG, 0x8E9EAEBE);
    WRITE_PERI_REG(SYSCON_SARADC_SAR2_PATT_TAB4_REG, 0xCEDEEEFE);

    SET_PERI_REG_BITS(SENS_SAR_MEAS_WAIT2_REG, SENS_FORCE_XPD_SAR, 3, SENS_FORCE_XPD_SAR_S);

    /* 1: SAR ADC2 controlled by DIG ADC2 CTRL or PWDET CTRL, 0: SAR ADC2 controlled by RTC ADC2 CTRL (R/W) */
    SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DIG_FORCE);
    /* 1: SAR ADC2 is controlled by DIG ADC2 CTRL, 0: SAR ADC2 is controlled by PWDET CTRL. */
    SET_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR2_MUX);

    /* Scan mode */
    /* 0: single mode, 1: double mode, 2: alternate mode. (R/W) */
    SET_PERI_REG_BITS(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_WORK_MODE, 0, SYSCON_SARADC_WORK_MODE_S);

    /* Selecting ADC for scanning in single mode */
    /* 0: SAR1, 1: SAR2, this setting is applicable in the single SAR mode. (R/W) */
    SET_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR_SEL);

    /* 1: sar_sel will be coded by the MSB of the 16-bit output data, in
     * this case, the resolution should not contain more than 11 bits;.
     * 0: using 12-bit SAR ADC resolution. (R/W) */
    CLEAR_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_DATA_SAR_SEL);

    /* SARADC_DATA_TO_I2S 1: I2S input data is from SAR ADC (for DMA), 0: I2S input data is from GPIO matrix. (R/W) */
    SET_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_DATA_TO_I2S);

    /* Max conversion number. (R/W) */
    SET_PERI_REG_BITS(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_MAX_MEAS_NUM, 255, SYSCON_SARADC_MAX_MEAS_NUM_S);
    /* Reserved. Please initialize to 0b1 (R/W) */
    SET_PERI_REG_BITS(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_MEAS_NUM_LIMIT, 0b1, SYSCON_SARADC_MEAS_NUM_LIMIT_S);

    /* 1: data to DIG ADC2 CTRL is inverted, 0: data is not inverted. (R/W) */
    SET_PERI_REG_MASK(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_SAR2_INV);

    /* Invert SAR ADC2 data. (R/W) *** NOT WORKING !!!!
    SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV); */
}

[image]

/*  NOT WORKED function  */
void sar1_scan_init(void)
{
    /* SAR ADC1, 0 - 15 means pattern table length of 1 - 16. (R/W)
     *  SET_PERI_REG_BITS(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR1_PATT_LEN, 0, SYSCON_SARADC_SAR1_PATT_LEN_S);
     *  Test pattern configuration byte 0xAD:
     * --[7:4] channel_sel: 10-->en_test
     * --[3:2] bit_width  : 3-->12bit
     * --[1:0] atten      : 1-->3dB attenuation */
    WRITE_PERI_REG(SYSCON_SARADC_SAR1_PATT_TAB1_REG, 0x1E1E2E3E);
    WRITE_PERI_REG(SYSCON_SARADC_SAR1_PATT_TAB2_REG, 0x4E5E6E7E);
    WRITE_PERI_REG(SYSCON_SARADC_SAR1_PATT_TAB3_REG, 0x8E9EAEBE);
    WRITE_PERI_REG(SYSCON_SARADC_SAR1_PATT_TAB4_REG, 0xCEDEEEFE);

    SET_PERI_REG_BITS(SENS_SAR_MEAS_WAIT2_REG, SENS_FORCE_XPD_SAR, 3, SENS_FORCE_XPD_SAR_S);

    /* 1: SAR ADC1 controlled by DIG ADC1 CTR, 0: SAR ADC1 controlled by RTC ADC1 CTRL. (R/W) */
    SET_PERI_REG_MASK(SENS_SAR_READ_CTRL_REG, SENS_SAR1_DIG_FORCE);

    /* Scan mode */
    /* 0: single mode, 1: double mode, 2: alternate mode. (R/W) */
    SET_PERI_REG_BITS(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_WORK_MODE, 0, SYSCON_SARADC_WORK_MODE_S);

    /* Selecting ADC for scanning in single mode */
    /* 0: SAR1, 1: SAR2, this setting is applicable in the single SAR mode. (R/W) */
    CLEAR_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR_SEL);

    /* 1: sar_sel will be coded by the MSB of the 16-bit output data, in
     * this case, the resolution should not contain more than 11 bits;.
     * 0: using 12-bit SAR ADC resolution. (R/W) */
    CLEAR_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_DATA_SAR_SEL);

    /* SARADC_DATA_TO_I2S 1: I2S input data is from SAR ADC (for DMA), 0: I2S input data is from GPIO matrix. (R/W) */
    SET_PERI_REG_MASK(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_DATA_TO_I2S);

    /* Max conversion number. (R/W) */
    SET_PERI_REG_BITS(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_MAX_MEAS_NUM, 255, SYSCON_SARADC_MAX_MEAS_NUM_S);
    /* Reserved. Please initialize to 0b1 (R/W) */
    SET_PERI_REG_BITS(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_MEAS_NUM_LIMIT, 0b1, SYSCON_SARADC_MEAS_NUM_LIMIT_S);

    /* 1: data to DIG ADC1 CTRL is inverted, 0: data is not inverted. (R/W) */
    SET_PERI_REG_MASK(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_SAR1_INV);

    /* Invert SAR ADC1 data. (R/W) *** NOT WORKING !!!!
    SET_PERI_REG_MASK(SENS_SAR_READ_CTRL_REG, SENS_SAR1_DATA_INV); */
}
[image]

Yoshi20, 2017-12-01:
Same problem here: All ADC reads are always zero when running i2s_adc_dac example with the standard configurations (esp-idf: 02304ad83e0a5f4815789d581446fa3afdd017b9).
Will this be fixed anytime soon?
Did you (@phonec) succeed with your workaround? If yes, how exactly?
Are there other useful methods to read an analogue microphone with a defined sample rate?

phonec, 2017-12-01:
@Yoshi20: (esp-idf: 02304ad) this indicates a merge for freertos, please clarify the link.

At me it turned out only with ADC2. I'm trying to figure it out, issue about this theme: [#1333](https://github.com/espressif/esp-idf/issues/1333)

Other methods of connecting an analog microphone are via an external I2S audio codec for example: [NXP SGTL5000](https://github.com/IoTBits/ESP32_SGTL5000_driver/).

Splediferous, 2017-12-13:
Same problem here. Working program a few weeks ago with an older SDK now fails to read any values apart from zeros. Trouble-shot down to "i2s_pop_sample" returning zeros. Tried "i2s_read_bytes" with same result.

Splediferous, 2018-01-26:
Is there no movement on this issue?
My application is now complete and am waiting to swap out timer tick acquisition for DMA.

jictannu, 2018-02-10:
@phonec Would you like to tell me how you initialize the SAR ADC2?

phonec, 2018-02-10:
Please see source from function [bootloader_random_enable()](https://github.com/espressif/esp-idf/blob/22489d70214a5b7650ab197ffd6ab73e9c50a772/components/bootloader_support/src/bootloader_random.c#L63).  But when you turn on the WiFi, the scaning is broken because wifi driver use some ADC2 lines.

costaud, 2018-02-11:
The fix is under MR, will be available very soon.

FayeY, 2018-02-12:
Hi, the fix has been merged, please have a try with the latest IDF, and feel free to reopen if the problem persists. Thanks.

closed 2018-02-12 by FayeY

GianlucaLoi, 2019-08-27:
Hello @FayeY, 

I opened a new issue with this problem of zero readings(https://github.com/espressif/esp-idf/issues/3973). 
Could you check it and tell me how to solve it? 

Regards,
Gianluca.

GianlucaLoi, 2019-08-28:
Hello @Alvin1Zhang,

Thank you! If you need more code or information, don't hesitate to contact me!

Regards,
Gianluca.

Alvin1Zhang, 2019-08-28:
@GianlucaLoi Thanks for reporting, we will take a look into issue [#3973  ] .