espressif/esp-idf #1204: BT GATTS: How to tell when client subscribed to notifications?

rojer, 2017-10-31:
My application is a GATT server and it can send notifications in certain cases.
However, i would like to know if the client actually subscribed/registered to receive them, as i don't want to waste resources unless client is actually subscribed.
I don't see any GATTS event raised when client registers for notifications, so - is there any other way to tell?

negativekelvin, 2017-10-31:
It should come as ESP_GATTS_WRITE_EVT for CCCD

rojer, 2017-10-31:
i have handlers for all the GATTS events and notification reg/unreg do not seem trigger any of them at present.

negativekelvin, 2017-10-31:
A writable descriptor that allows client to configure notifications for that characteristic

rojer, 2017-10-31:
um... so, as a service, am i supposed to do anything special to receive those events?
notifications themselves work fine - when i send them (blindly), the clients receives them just fine.
but i don't get any GATTS events when client regs/unregs.

negativekelvin, 2017-10-31:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server/main/gatts_demo.c#L367

rojer, 2017-10-31:
that... is not obvious :)
but really, i should re-examine the examples again, looks like they've changed a lot.

but i'm definitely not seeing write events, i would've. weird. but i need to verify that my client (BLE Scanner app) actually sends them.

negativekelvin, 2017-10-31:
Client will not register if it doesn't discover CCCD

rojer, 2017-11-03:
figured it out, thanks!

closed 2017-11-03 by rojer