espressif/esp-idf #1205: External flash chain bootloader

OtherCrashOverride, 2017-10-31:
Rather than burning efuses, it is desirable to have a bootloader that can be written to WROOM32 module flash that will configure SPI for off-module flash and continue booting from there (disabling module flash).

This provides a benefit for prototyping designs since the WROOM32 does not need to be thrown away if the configuration changes.  This also provides benefit for production by offering configuration flexibility.

projectgus, 2017-10-31:
When you say "burning efuses", you mean for SPI flash pin remapping? So you want something like:

* ESP32 loads second stage bootloader from internal flash, boots it (ie same as "normal" ESP-IDF)
* Second stage bootloader reads configuration from somewhere (?) and optionally swaps flash pins before loading the app.

We don't support such a configuration but it is certainly possible with a custom second stage bootloader.

You should be able to call the following ROM function from the second stage loader to configure the pins (the ROM function doesn't anything particularly magic, just sets the GPIO matrix, but given it already exists you may as well use it):
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/rom/spi_flash.h#L515

The "spiconfig" param to this function has the pin numbers (apart from WP pin) encoded as per the results of this function:
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/rom/efuse.h#L62

But you don't need to read these from efuse, you can build the integer using the macros at the above link.

There may be some additional complexity because the first stage ROM bootloader has already initialised the SPI flash chip. None has probably tested this exact configuration (starting from one flash chip and then switching). But the above should get you most of the way there, if you get stuck then let me know.

Presumably once we have a fix for https://github.com/espressif/esptool/issues/243 this may also illuminate this (I'm not sure what's going on there, unfortunately it will probably be a couple of weeks before I'm in a position to test it properly).

Given this is more of a discussion than an "issue" in IDF then a good place to continue discussing it is probably the forums - https://esp32.com

OtherCrashOverride, 2017-10-31:
Yes, "burning efuses" refers to
```
SPI_PAD_CONFIG_CLK
SPI_PAD_CONFIG_Q
SPI_PAD_CONFIG_D
SPI_PAD_CONFIG_HD
SPI_PAD_CONFIG_CS0
```
My initial concept is to just #define the pins in code rather than acquiring the information.  I believe I can use the ROM functions to set the configuration.  However, the key is going to be what happens after configuration.  Ideally, I would like to just call a "warm boot" function in ROM that would use the now updated SPI configuration. 

I intended this to be more a supported "feature request" than a "discussion".  If I am to author it myself, then I agree its a "discussion" and better suited to the forum.

OtherCrashOverride, 2017-10-31:
> you can implement a hook here that reads some external configuration from somewhere and reconfigures SPI flash as needed before continuing the boot process). Will that help you achieve the goal you have?

That does indeed sound like a good solution.  Is the current bootloader code open source somewhere?

projectgus, 2017-10-31:
> That does indeed sound like a good solution.

Great!

> Is the current bootloader code open source somewhere?

Yes:
https://github.com/espressif/esp-idf/tree/master/components/bootloader

A lot of the related functionality is also in the IDF bootloader_support component (which contains code which can be compiled either for the bootloader or for inclusion in an ESP-IDF app).

Note this is the second stage bootloader which is read from offset 0x1000 in flash. The initial bootloader in ROM initialises SPI flash based on efuse, loads this bootloader into RAM, and then executes it. But there's no reason this bootloader should not be able to re-initialise SPI flash differently before continuing.

The ESP-IDF boot process is explained in detail here:
https://esp-idf.readthedocs.io/en/latest/api-guides/general-notes.html#application-startup-flow

OtherCrashOverride, 2017-11-01:
I have started testing reconfiguring the flash using the `esp_rom_spiflash_attach` ROM API call.  However, I am see what I believe is the same behavior as the esptool.py issue (espressif/esptool#243) I encountered.  The configuration is ignored and the internal flash is used.  Just as in the other issue, changing the CS does not affect it but changing other numbers (like Q) does.

OtherCrashOverride, 2017-11-01:
I should probably explicitly state my concern is that burning the efuses to this configuration also will **not** work since it seems to be a ROM code issue.  This would render any ESP32 with this configuration a "brick".

OtherCrashOverride, 2017-11-02:
I have settled on the following solution:

Rather than calling ROM routines, I am directly manipulating the GPIO configuration:
```
    // Route CS to external flash
    gpio_matrix_out(16, SPICS0_OUT_IDX, 0, 0);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_CMD_U, 2);

    // Disable CS on module flash
    gpio_pad_select_gpio(11);
    GPIO_OUTPUT_SET(11, 1);
```
The above lines are placed in the bootloader just before `esp_rom_spiflash_unlock();`:
https://github.com/espressif/esp-idf/blob/80eb3b6ddbc218bcce830771730f902480007578/components/bootloader/subproject/main/bootloader_start.c#L470

This causes the rest of initialization to use the external flash.  It is testing and working with my application.  The reported configuration is 80Mhz, QIO, 16MB.  The "SLOW READ" issue was resolved by re-wiring.

With the above, boot starts on the WROOM32 4MB flash and continues on a 16MB external flash.  The '--no-stub' flag is still required for esptool.py to write to the external flash.

Buckler89, 2018-03-28:
@OtherCrashOverride  
I have a similar situation: my 16MB flash is connected as yours except for CS pin that is IO5 in my case (see also [this issue.](https://github.com/espressif/esptool/issues/243)
After your bootloader modification:

```
    // Route CS to external flash
    gpio_matrix_out(5, SPICS0_OUT_IDX, 0, 0);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_CMD_U, 2);
    // Disable CS on module flash
    gpio_pad_select_gpio(11);
    GPIO_OUTPUT_SET(11, 1);
```

I flash the ESP with "make flash" command. After that when serial monitor starts, the log shows me the message:
```
I (28) boot: ESP-IDF v3.1-dev-463-g77eae33a-dirty 2nd stage bootloader
I (28) boot: compile time 17:04:49
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 80MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot: End of partition table
E (62) boot: No bootable app partitions in the partition table
user code done
```

2 stuff here:

1.  My flash is a 16 MB flash, not 2 MB as shown. Any ideas about why this happens?

2. apart from the flash size detection, I think that the error is right because I haven't' yet upload nothing to the 16 MB flash. But how can I do that? With your solution, what is the right command to write all binary image? I assume that I need to upload just partition table and app image to the external flash ( neither bootloader image because it is already uploaded to 4MB flash). 

Thank you

OtherCrashOverride, 2018-04-06:
> what is the right command to write all binary image?

I documented it in the issue referenced.  Adding "--no-stub" to the esptool.py command allows flashing the external chip. I believe that 'make' outputs the commands required to flash your project's data.  Simply add the "--no-stub" and flash chip options to the commands shown by it.

Note that GPIO5 is in a different power domain (VIO) than GPIO16 (VSDIO).  Ensure your flash chip is compatible with esp-idf for detection and using the correct power.

For testing, you may wish to wire /CS to IO16 since its known to work and the procedure is documented.  After verifying operation, you can move it to the IO pin of your choice to isolate the issue.

Buckler89, 2018-06-14:
Hello, thanks for the reply.
I finally managed to work on this project again in the last days. Adding the --no-stub and --spi-configuration to the command shown after the compilation phase seems to work fine (it works both with the CS connected to gpio5 16).
This is the output:
```
Connecting...
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Configuring SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 16MB
Erasing flash...
Compressed 24624 bytes to 13900...
Took 0.38s to erase flash block
Wrote 24624 bytes (13900 compressed) at 0x00001000 in 1.4 seconds (effective 141.3 kbit/s)...
Hash of data verified.
Erasing flash...
Compressed 930560 bytes to 528185...
Took 2.70s to erase flash block
Wrote 930560 bytes (528185 compressed) at 0x00010000 in 52.2 seconds (effective 142.6 kbit/s)...
Hash of data verified.
Erasing flash...
Compressed 3072 bytes to 126...
Took 0.06s to erase flash block
Wrote 3072 bytes (126 compressed) at 0x00008000 in 0.1 seconds (effective 253.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
As you can see the auto-detect flash size is correct: 16MB

The problem, however, is that I can not boot from the external memory: I have applied the patch proposed by you above, but not functional. I always see the data stored on the 4M flash instead of the 16MB flash.

In summary, I made 2 steps:

1. change to the bootloader (with the progress of the master, I made the change in the bootloader_init.c file, just above `esp_rom_spiflash_unlock ();` as you explained.
2.  flash with --no-stub and --spi-configuration parameter

Did I forget something?

Do you know if in the last few months there have been additions of functions to facilitate this custom setting?

Thank you

negativekelvin, 2018-06-14:
Did you flash the modified bootloader to internal flash (without --no-stub and --spi-configuration)?

Buckler89, 2018-06-19:
Thanks, I had not done it now. However it still does not work. In particular I flasked the bootloader (with the patch) in the 4MB flash, then I took the command sugerito at the end of the compilation and I added --no-stub and --spi-configuration. So I loaded the firmware into the 16MB flash (the logs confirm it). All this by repeating the addresses of the partitions. Then the partiton.bin at the address 0x8000 and the app at the address 0x10000 of the flash 16MB.

I also modified the bootloader to print the flash size detected before and after changing the GPIO configuration.

   ```
    // Route CS to external flash
    gpio_matrix_out(16, SPICS0_OUT_IDX, 0, 0);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_CMD_U, 2);

    // Disable CS on module flash
    gpio_pad_select_gpio(11);
    GPIO_OUTPUT_SET(11, 1);

    esp_image_header_t fhdr2;
    bootloader_flash_read(ESP_BOOTLOADER_OFFSET, &fhdr2, sizeof(esp_image_header_t), true);
    print_flash_info(&fhdr2);

```
When I start the card it tells me 2 times that the flash is 4MB. So the modification of the GPIO seems to be ignored.

What may be the problem now?

negativekelvin, 2018-06-19:
The flash size info from the header is based on a value set in menuconfig not based on reading the hardware registers of the flash chip

Buckler89, 2018-06-19:
Are you sure? because in the menuconfig I set the flash 16MB, but during the boot it prints 2 times that the flash is 4MB

negativekelvin, 2018-06-19:
Well if you erase the internal flash and only have a bootloader there and your app runs then you know it is running from external

Buckler89, 2018-07-02:
Hi Guys, I'm here again yet without success.
I summarize all the step I've done:

This is my custom partition table:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,   0x6000,
phy_init, data, phy,     0xf000,   0x1000,
factory,  app,  factory, 0x10000,  1M,
storage1,  data, spiffs,  0x110000, 1M,
storage2,  data, spiffs,  0x210000, 0x110000,
```
(As you can see at the moment I do not exploit all the 16MB, just because I'm trying to make the code work as it is in the external flash. This should not create problems.)

This is the bootloader patch in the **bootloader_init.c** after line 155: 

```
    // Route CS to external flash
    gpio_matrix_out(5, SPICS0_OUT_IDX, 0, 0); // devkit pin
    // gpio_matrix_out(16, SPICS0_OUT_IDX, 0, 0); // tx183 pin
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_CMD_U, 2);

    // Disable CS on module flash
    gpio_pad_select_gpio(11);
    GPIO_OUTPUT_SET(11, 1);
    esp_image_header_t fhdr2;
    bootloader_flash_read(ESP_BOOTLOADER_OFFSET, &fhdr2, sizeof(esp_image_header_t), true);
    print_flash_info(&fhdr2);
```

I set the `flash_size` in menuconfig to 16MB. I left the parameter `Detect flash size when flashing bootloader`  to _yes_ (but i tried also the whole procedure described her with this parameter set to _no_)

I compiled the code (bootloader patched) with `make` command. 

Then I flashed  the bootloader only in the internal 4 MB flash at 0x1000 with the following command (without --no-stub and --spi-configuration):

```
python myhome/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 myhome/myproject/esp32-firmware/build/bootloader/bootloader.bin
esptool.py v2.3.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 24640 bytes to 13920...
Wrote 24640 bytes (13920 compressed) at 0x00001000 in 1.2 seconds (effective 160.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

After that i flashed the external 16MB flash with the partition table (at 0x8000) and firmware(at 0x10000) with the following commands:
```
python /home/buckler/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --no-stub --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset  write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect --spi-connection 6,7,8,9,5 0x10000 /media/buckler/Data/Phd/Project/Proietti/AsaPlastici/esp32-firmware/build/ESplug-firmware.bin 0x8000 /media/buckler/Data/Phd/Project/Proietti/AsaPlastici/esp32-firmware/build/partitions.binesptool.py v2.3.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Configuring SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 16MB
Erasing flash...
Compressed 930896 bytes to 528308...
Took 2.72s to erase flash block
Wrote 930896 bytes (528308 compressed) at 0x00010000 in 52.2 seconds (effective 142.6 kbit/s)...
Hash of data verified.
[... 100 lines cut ...]
D (118) boot: load partition table entry 0x3f408000
D (123) boot: type=1 subtype=2
I (126) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (134) boot: load partition table entry 0x3f408020
D (139) boot: type=1 subtype=1
I (142) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (150) boot: load partition table entry 0x3f408040
D (155) boot: type=0 subtype=0
I (158) boot:  2 factory          factory app      00 00 00010000 00100000
D (165) boot: load partition table entry 0x3f408060
D (170) boot: type=1 subtype=82
I (173) boot:  3 storage1         Unknown data     01 82 00110000 00100000
D (181) boot: load partition table entry 0x3f408080
D (186) boot: type=1 subtype=82
I (189) boot:  4 storage2         Unknown data     01 82 00210000 00110000
I (197) boot: End of partition table
D (201) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (207) esp_image: reading image header @ 0x10000
...
...etc (start loading partitions)
```

Any suggestions?

Thanks!

negativekelvin, 2018-07-02:
If bootloader_flash_read is reading a valid header and there is no bootloader on the external flash then you are still using internal flash.

Buckler89, 2018-07-09:
@negativekelvin Yes, in fact I expect that during the reading of the bootloader header after the change of the CS pins for the flash memory,  it will give me an error.

In your opinion, are the steps described in the previous post correct?
What other problem could there be?

Other insights:
I think the problem is that the bootloader patch is not working properly. @OtherCrashOverride Can you give me a hint?  
It seems that the bootloader patch activates pin 5 but can not disable pin 11, keeping both flashes active at the same time.
I say this for the following reason:
loading in the 4MB flash all the code with `make flash` (patched bootloader + partition table + all partitions), and nothing in the 16MB flash, I get the following result:

```
$ make monitor 
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8084
load:0x40078000,len:0
load:0x40078000,len:16456
entry 0x400786f8
D (71) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (35) boot: ESP-IDF v3.1-dev-1344-g7aa7b35a-dirty 2nd stage bootloader
I (35) boot: compile time 10:31:36
D (36) boot: magic e9
D (38) boot: segments 04
D (41) boot: spi_mode 02
D (43) boot: spi_speed 00
D (46) boot: spi_size 04
I (49) boot: SPI Speed      : 40MHz
I (53) boot: SPI Mode       : DIO
I (57) boot: SPI Flash Size : 16MB                                                    <---------- after pin reconfig
I (61) boot: Enabling RNG early entropy source...
D (66) boot: magic e9
D (69) boot: segments 04
D (71) boot: spi_mode 02
D (74) boot: spi_speed 00
D (77) boot: spi_size 04
I (79) boot: SPI Speed      : 40MHz
I (83) boot: SPI Mode       : DIO
I (87) boot: SPI Flash Size : 16MB                                                    <---------- before pin reconfig
D (91) bootloader_flash: mmu set paddr=00000000 count=1
D (97) boot: mapped partition table 0x8000 at 0x3f408000
E (102) flash_parts: partition 3 invalid magic number 0x50a2
E (108) boot: Failed to verify partition table
E (114) boot: load partition table error!
user code done
```
Note the double print of 16MB

If instead load the partition table also in the 16MB flash, the application starts correctly, but the booting takes place in the 4MB flash, because in the 16MB there is only the partition table.

Do you have any other suggestions on how to modify the bootloader in order to start it from the 4MB flash and then load the partition table and the various partitions on the 16MB external flash?

I attach the modified bootloader_init.c file:
[bootloader_init.zip](https://github.com/espressif/esp-idf/files/2175606/bootloader_init.zip)

I'm going crazy ... please ... help me XD

projectgus, 2019-03-01:
@Buckler89 I don't know if you're still stuck on this, but it may be worth opening a thread on https://esp32.com as this has become more of a discussion topic ("how to modify IDF bootloader to achieve X") rather than an issue in IDF that we can fix directly.

closed 2019-03-01 by projectgus