espressif/esp-idf #1208: wifi stops working after a while

codeib, 2017-11-01:
I have a problem with the esp32  , after initialising the module I use wifi to transmit and receive udp packets but this stops working after some time . 

 I use the lwip udp library to transmit 31 bytes and receive 562 bytes of data each time , the data transmission only works 10 times with the wifi rx buffer set to 10. The wifi remains connected but all data transmission stops  . If I increase the wifi rx dynamic buffer I can increase number of packets transmitted. 

When the data transmission stop if the wifi is forced to disconnect , re-association with the access point also fails . The only way to restore this is to restart the module.

Wifi DisconnectedI (302899) wifi: n:2 1, o:3 0, ap:255 255, sta:2 1, prof:1
I (302899) wifi: state: init -> auth (b0)
I (302899) wifi: state: auth -> assoc (0)
I (302909) wifi: state: assoc -> run (10)
I (305909) wifi: pm start, type:0
I (305919) wifi: state: run -> auth (2a0)
I (305919) wifi: pm stop, total sleep time: 0/10803

I think this has something to do with release of memory , is it possible to force the wifi memory to be flushed and is there a fix for this issue.

kakegu, 2017-11-03:
I have this problem too. Have you solved it?If it's done, please let me know

codeib, 2017-11-07:
I fixed this issue by releasing the memory allocated to p buffer from within the recv call back function . This is mentioned within the lwip udp source.

closed 2017-11-07 by codeib

vahidmousavi, 2020-10-13:
> I fixed this issue by releasing the memory allocated to p buffer from within the recv call back function . This is mentioned within the lwip udp source.

Hello, I wish I could contact you and ask about the way of implementing and fixing this problem.