espressif/esp-idf #1210: Timer abort using function touch_pad_filter_start(10)

bmetman, 2017-11-01:
Dear Espressif team, 

I'm getting the error below when using the **touch_pad_filter_start()** function as found in the touchpad examples. 
```
/home/user/esp/esp-idf/components/freertos/./timers.c:670 (prvProcessReceivedCommands)- assert failed!
abort() was called at PC 0x40088bb7 on core 0
0x40088bb7: prvProcessReceivedCommands at /home/user/esp/esp-idf/components/freertos/./timers.c:839

Backtrace: 0x40089a90:0x3ffcf470 0x40089b8f:0x3ffcf490 0x40088bb7:0x3ffcf4b0 0x40088c17:0x3ffcf4f0
0x40089a90: invoke_abort at /home/user/esp/esp-idf/components/esp32/./panic.c:553

0x40089b8f: abort at /home/user/esp/esp-idf/components/esp32/./panic.c:553

0x40088bb7: prvProcessReceivedCommands at /home/user/esp/esp-idf/components/freertos/./timers.c:839

0x40088c17: prvTimerTask at /home/user/esp/esp-idf/components/freertos/./timers.c:839
```
My guess is that because the timer that is used for the touch_pad_filter_start function has priority 1, it conflicts with another priority 1 task. However, all my own custom tasks I launch with a priority of 2 or lower. Functions which do probably run at priority 1 are the callback functions I've registered for BLE integration:
```
  esp_ble_gatts_register_callback(gatts_event_handler);
  esp_ble_gap_register_callback(gap_event_handler);
```
In fact, when I don't start up the BLE component, the touchpad filter works fine, so I'm fairly sure the problem is with these tasks running at the same priority. 

I've referenced this problem before in the issue found [here](https://github.com/espressif/esp-idf/issues/1143). The problem is that the priority of these BLE callback functions can't be changed. 

Do you know how I may solve this issue? I've tried starting the touchpad filter before starting the BLE callbacks, but it seems even with significant delays and semaphore passing between tasks that the same error keeps occurring. 

Thank you in advance for your help.

bmetman, 2017-11-21:
Hey there, any indication on when you have time look into this issue?

costaud, 2018-01-29:
Really sorry for the late reply, would you please provide some of the test code along with the sdkconfig file? In addition, would you also kindly provide the idf commit ID you are using? 

Seems we can reproduce this issue on master branch right now.