espressif/esp-idf #1213: ESP32 BLE characteristics

Goutham8128, 2017-11-02:
Hi

I executed the ESP 32 BLE gateway code using arduino IDE.I didnt get any errors.In my case i am using a BLE sensor dongle and i want the ESP 32 module to retrieve the services and characteristics of the BLE sensor dongle which i am using.The BLE sensor dongle has a temperature,pressure and Accelerometer sensor.Each sensor has specific UUID's. When i open the serial monitor after executing the code the serial monitor is not displaying the sensor data.I am not getting sensor data.

nkolban, 2017-11-10:
I don't think there is enough information in the report to provide good assistance.  Perhaps if you pasted your Arduino script to a pastebin and posted a link to it?

Goutham8128, 2017-11-10:
Hi

Sorry for not giving the information correctly.I am using a sensor dongle which has sensor information in it. It is acting as a BLE server. I am using ESP32 acting as a BLE client module.I upload the BLE client code into the esp32 but after its execution it is just displaying the BLE server name and its service UUID. It is not able to read the characteristics of the BLE server. I want to read the characteristics of the BLE server in the serial monitor. Can u help me out with this?

Yulong-espressif, 2017-12-15:
@Goutham8128 You can try our BLE server demo to connect to the BLE client demo first. Thanks.

Goutham8128, 2017-12-15:
Thanks for the response. Where can I find the one which you are referring?

FayeY, 2017-12-21:
Could those [examples](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth) in IDF help?

Goutham8128, 2017-12-21:
No

The code is running but the sensor which i am using as a BLE server is not being received by the BLE client i.e Esp32. I couldn't find any source for this.

chegewara, 2017-12-21:
What is this sensor? Maybe it would be better to ask on forum or forum related to this sensor?

Goutham8128, 2017-12-21:
The one which we discussed previously which Mr. Kolban has bought.

FayeY, 2017-12-21:
Sorry that maybe I misunderstood it. It seems to be a problem of the sensor. 
How about using a mobile phone as a BLE client? 
Could a mobile phone read the characteristics of your sensor?

Goutham8128, 2017-12-21:
U mean that the Server code running on DA14583 which is advertising the data should be modified?.
If the services of the module are read by nrfconnect then why the client program running on esp32 is not able to read the services?

Weijian-Espressif, 2018-03-14:
@Goutham8128 Can you use lightblue APP or nrf connect APP to test the sensor,  do the APP work well?

FayeY, 2018-04-02:
Closing due to lack of response, please feel free to reopen if this is still an issue you can supply more details. Thanks.

closed 2018-04-02 by FayeY