espressif/esp-idf #1214: Error on io_mux_reg.h

urs8000, 2017-11-02:
platform: Win10, Arduino 1.8.5, EPS32 new installed today
**when** compiling "blink" I get a lot of errors**
In file included from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/soc.h:20:0,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32/rom/ets_sys.h:21,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:21,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-log.h:111,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal.h:49,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-adc.h:27,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-adc.c:15:
C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/io_mux_reg.h: In function 'PIN_PULLUP_DIS':
C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/soc.h:142:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));

reading a comment in io_mux_reg
 *     The PIN_PULL[UP|DWN]_[EN|DIS]() functions used to exist as macros in previous SDK versions. 
 *     Unfortunately, however, they do not work for some GPIOs on the ESP32 chip, which needs pullups 
 *     and -downs turned on and off through RTC registers. The functions still exist for compatibility 
 *     with older code, but are marked as deprecated in order to generate a warning.
it should only give warnings.
what's the problem and how could I solve it?    thanks!

urs8000, 2017-11-02:
schould be in the path arduino-esp32.  sorry

closed 2017-11-02 by urs8000

davidm-public, 2018-02-27:
same problem
```
rs/davem/Documents/Arduino/hardware/espressif/arduino-esp32/tools/sdk/include/soc/soc/io_mux_reg.h: In function 'PIN_PULLUP_DIS':
/Users/davem/Documents/Arduino/hardware/espressif/arduino-esp32/tools/sdk/include/esp32/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
```