espressif/esp-idf #1215: Separate component directory for drivers of devices like sensors

gschorcht, 2017-11-02:
Are there any plans to introduce a separate directory where users can contribute their drivers for external devices like sensors that are very often used which are not part of the core idf but help to start new projects? Or should external device drivers kept left in separate user projects? 

For example in [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos.git) for ESP8266 there is an ```extra``` directory for that purpose which helps a lot to start new projects?

chanibal, 2017-11-02:
This sounds like https://github.com/espressif/esp-iot-solution

igrr, 2017-11-02:
We have some plans for an ESP-IDF package (component) manager, which should facilitate dependency tracking and installation of components developed by Espressif and third parties. 

Until we have it, the suggested approach is to publish such components on Github, and create a topic in the "Showcase" section of ESP32.com forum with a brief introduction about the component.

closed 2017-11-08 by igrr