espressif/esp-idf #1218: COAP experiment between two esp32 as server/client

singhp6, 2017-11-02:
I am trying the coap experiment between two esp32 as server/client and I am able to connect both esp32(server and client) to wifi access point. However, I am not sure what 'URI' exactly (or in which syntax) should I use to get the coap communication going between the two esp32. What exactly should the "Target uri" be for two boards to start communicating over wifi (for coap experiment)? I am using the syntax at client as 'coap://ip address(of server)' in "make menuconfig --> example configuration"...For example coap://10.13.215.134/ (see 1.PNG) but getting an error 'DNS lookup failed' (see 2.PNG). I am not so sure if thats the right way of doing it. Am I doing anything wrong here? I tried different URI's to see if it works or not, like the "coap://californium.eclipse.org/" or "coap://coap.me/", but could get anything with this. I got the same error.

[image]
[image]

0xff007, 2017-11-04:
the coap experiment between esp32 client and libcoap-server.
[image]
[image]
and I try to capture with WireShark.
[image]

reset esp32 Restart multiple times,
[image]
[image]

I guess it's response processing error. 
I don't know how to solve it.):

singhp6, 2017-11-04:
Did you make one esp32 as a server? What is the libcoap server that you have? Also, how did you run both server and client? I used "make flash monitor". I want to know what is the "URI" and in what format we should be using it.

singhp6, 2017-11-04:
Questions:

1) Did you make one esp32 as a server? What is the libcoap server that you have?
2) How did you run both server and client? I used "make flash monitor". It seems you ran libcoap server a bit differently.
3) I want to know what is the "URI" and in what format we should be using it.

0xff007, 2017-11-05:
@singhp6 
1. I made my pc as a server by running the example server of libcoap 。
2. I made one esp32 as  a client, url is coap://10,.1.1.129 , 
as you see，libcoapserver register resouce “NULL" , then, send response to client with payload INDEX 
[image]
[image]

Making sure that runing a coap server correct ，then ， making one esp32 as a client。
 3. I test URI in format ”coap://serverdomainorip/path" 

in a word , i think it is response processing error.
now ,i'm reviewing the code of [libcoap](https://github.com/obgm/libcoap) and the example of [coap-client](https://github.com/espressif/esp-idf/blob/master/examples/protocols/coap_client/main/coap_client_example_main.c).

next ,i'll try to make one esp32 as server , use my pc send request .

I hope I can find a solution.

zhang-jun-hao, 2017-11-15:
Hi @singhp6  @0xff007 

The sever url is coap://x.x.x.x/Espressif

Please change the line code:
    result = select( FD_SETSIZE, &readfds, 0, 0, &tv ); 
to:  
    result = select( ctx->sockfd+1, &readfds, 0, 0, &tv );
in file coap_client_example_main.c and coap_server_example_main.c.

Or use the attached demo file:
[coap_client.tar.gz](https://github.com/espressif/esp-idf/files/1473616/coap_client.tar.gz)
[coap_server.tar.gz](https://github.com/espressif/esp-idf/files/1473617/coap_server.tar.gz)

Thanks!

closed 2017-11-21 by FayeY

singhp6, 2017-12-07:
@0xff007 Hi,

Did you get the solution for the problem which you were looking for (above)? If yes, could you please explain what did you do? are you now not getting the response processing error? Was there a problem with the URI that you were taking?