espressif/esp-idf #1220: SPI SLAVE receives data instability

L-eeX, 2017-11-03:
The data received by the spi slave is very unstable.I used the spi slave development project, and I've been plagued by this problem for half a month.I want to know when I can fix it?

yake827, 2017-11-03:
Hi @L-eeX , Can you describe your specific phenomenon for us to analysis

L-eeX, 2017-11-04:
@yake827 Download the example
**# Slave machine prints information：**
Received: ゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ?
Received: ゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ?
Received: ゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ?
Received: ゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ?
**# Host printing information**
Received: 
Received: 
Received: 
Received:

L-eeX, 2017-11-04:
@yake827 This is my own code
# Slave machine prints information：
Received: 11 11 11 11
ret: 0
Received: 11 0 0 0
ret: 0
Received: 11 0 0 0
ret: 0
Received: 11 11 11 11
ret: 0
Received: 11 11 11 11
# Host printing information
Received: 42 170 170 170
Received: 42 170 170 170
Received: 42 170 170 170
Received: 42 170 170 170
Received: 11 11 11 11
Received: 41 41 0 200
Received: 42 170 170 170
Received: 42 169 82 16

yake827, 2017-11-09:
Hi @L-eeX , it seems happen signal loss, I suggest you solder the SPI pins of the master and slave to each other using shorter cables.

L-eeX, 2017-11-09:
@yake827
I use the dupont line to connect two devices, which should not be a line problem

yake827, 2017-11-09:
@L-eeX I also test it using the dupont lines firstly, I found when the line is too long , the signal will loss frequently.So shorter dupont line maybe completely eradicate the signal loss.

L-eeX, 2017-11-09:
Hi @yake827 ,I have just used 10 cm (3.94 inch) dupont line, which should not influence the IIC.

yake827, 2017-11-09:
@L-eeX Can you tell me your IDF commit? I will test it again.

L-eeX, 2017-11-09:
# receiver
`/* SPI Slave example, receiver (uses SPI Slave driver to communicate with sender)

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"
#include "lwip/igmp.h"

#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "soc/rtc_cntl_reg.h"
#include "rom/cache.h"
#include "driver/spi_slave.h"
#include "esp_log.h"
#include "esp_spi_flash.h"

/*
SPI receiver (slave) example.

This example is supposed to work together with the SPI sender. It uses the standard SPI pins (MISO, MOSI, SCLK, CS) to 
transmit data over in a full-duplex fashion, that is, while the master puts data on the MOSI pin, the slave puts its own
data on the MISO pin.

This example uses one extra pin: GPIO_HANDSHAKE is used as a handshake pin. After a transmission has been set up and we're
ready to send/receive data, this code uses a callback to set the handshake pin high. The sender will detect this and start
sending a transaction. As soon as the transaction is done, the line gets set low again.
*/

/*
Pins in use. The SPI Master can use the GPIO mux, so feel free to change these if needed.
*/
//#define GPIO_HANDSHAKE 2
#define GPIO_MOSI 12
#define GPIO_MISO 13
#define GPIO_SCLK 15
#define GPIO_CS 14

/*
//Called after a transaction is queued and ready for pickup by master. We use this to set the handshake line high.
void my_post_setup_cb(spi_slave_transaction_t *trans) {
    WRITE_PERI_REG(GPIO_OUT_W1TS_REG, (1<<GPIO_HANDSHAKE));
}

//Called after transaction is sent/received. We use this to set the handshake line low.
void my_post_trans_cb(spi_slave_transaction_t *trans) {
    WRITE_PERI_REG(GPIO_OUT_W1TC_REG, (1<<GPIO_HANDSHAKE));
}
*/

//Main application
void app_main()
[... 225 lines cut ...]
    //Initialize the SPI bus and add the device we want to send stuff to.
    ret=spi_bus_initialize(HSPI_HOST, &buscfg, 1);
    assert(ret==ESP_OK);
    ret=spi_bus_add_device(HSPI_HOST, &devcfg, &handle);
    assert(ret==ESP_OK);

    //Assume the slave is ready for the first transmission: if the slave started up before us, we will not detect 
    //positive edge on the handshake line.
    //xSemaphoreGive(rdySem);

    while(1) {
        //snprintf(sendbuf, 128, "Sender, transmission no. %04i. Last time, I received: \"%s\"", n, recvbuf);
        t.length=4*8; //128 bytes
        t.tx_buffer=sendbuf;
        t.rx_buffer=recvbuf;
        //Wait for slave to be ready for next byte before sending
        //xSemaphoreTake(rdySem, 100); //Wait until slave is ready);//portMAX_DELAY); //Wait until slave is ready
        ret=spi_device_transmit(handle, &t);
        printf("Received: %d %d %d %d\n",recvbuf[0],recvbuf[1],recvbuf[2],recvbuf[3]);
    }

    //Never reached.
    ret=spi_bus_remove_device(handle);
    assert(ret==ESP_OK);
}`

yake827, 2017-11-09:
Hi @L-eeX , Because you using the 10 cm (3.94 inch) dupont line, I suggest the  clock_speed_hz  less than 3MHz.

L-eeX, 2017-11-09:
Hi @ yake827，Almost all the frequencies have been tried, and the data cannot be fully received

yake827, 2017-11-09:
Hi @L-eeX , could you  give me your ESP-IDF commit? I have tested the SPI slave demo using the latest commit and find it is OK.

L-eeX, 2017-11-09:
Hi @ L-eeX，I downloaded the latest esp-idf，I just changed the spi_slave example，My spi slave code is in the reply above。

FayeY, 2018-01-05:
Hi @L-eeX , sorry that we cannot reproduce your problem, could you check your hardware? We think that maybe it is a hardware issue.

closed 2018-01-11 by FayeY

9599, 2020-08-08:
Do you solve the problem. I have the same problem on the example of spi_slave.