espressif/esp-idf #1222: RMT boot loop after unrelated crash

chanibal, 2017-11-03:
Sometimes after a supposedly unrelated to RMT crash, I get a RMT boot loop.

```
gpio_install_isr_service(0)
// E (2858) gpio: gpio_install_isr_service(406): GPIO isr service already installed

ESP_ERROR_CHECK(rmt_config(&rmt));
// D (2888) rmt: Rmt Tx Channel 0|Gpio 5|Sclk_Hz 80000000|Div 80|Carrier_Hz 10000|Duty 0

ESP_ERROR_CHECK(rmt_driver_install(channel, 0, 0));
// ?[V (2908) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
// V (2918) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0xE
// 🢂🢂🢂 D (2918) rmt: RMT INTR : TX END 🢀🢀🢀
// Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
// Register dump:
// PC      : 0x4008470f  PS      : 0x00060031  A0      : 0x40082ba8  A1      : 0x3ffc0be0
// 0x4008470f: rmt_driver_isr_default at (...)/esp-idf/components/driver/rmt.c:780
// 
// 0x40082ba8: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105
// 
// A2      : 0x0000000f  A3      : 0x00008000  A4      : 0x00000005  A5      : 0x00000000
// A6      : 0x00000000  A7      : 0x3ffdd278  A8      : 0x8008470c  A9      : 0x3ffc0b50
// A10     : 0x00000024  A11     : 0x3ffc0be0  A12     : 0x0000000c  A13     : 0xffffffff
// A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
// EXCVADDR: 0x00000014  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
// 
// Backtrace: 0x4008470f:0x3ffc0be0 0x40082ba5:0x3ffc0c10 0x4000bfed:0x00000000
// 0x4008470f: rmt_driver_isr_default at (...)/esp-idf/components/driver/rmt.c:780
// 
// 0x40082ba5: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105
// 
// 
// Rebooting...
```

The interesting part is `D (2918) rmt: RMT INTR : TX END` (marked in log above) - this means that a RMT has just ended... but this is just after a restart and init. RMT was used before that, so it's an echo of something that happened before booting and should not be executed.

At first, I thought it might be related to my patch #1174, but this does not seem to be the case. The boot loop happens in the init phase of my firmware, just after an wifi access point has been created and a few tasks start running.

The crash that causes it is probably some pointery memory corruption, still not found.

Resetting by `idf-monitor` helps.

chanibal, 2017-11-03:
Using  ESP-IDF v3.0-dev-1057-g4048db35

chanibal, 2017-11-08:
Some time ago, I found the unrelated crash. It was very unrelated, `ESP_LOGI` of a object that had a C string, but the object was `delete`d before printing.

RMT boot loop still happens when ESP dies sometimes. Any idea on how to fix this?

koobest, 2017-11-29:
hi, @ chanibal , would you kindly provide a simple code, we need to reproduce the problem  and find the cause.

chanibal, 2017-11-30:
Hi, cannot reproduce crash at this moment, but can reproduce the premature `TX END` before anything has been sent. This is the code that displays this issue:

```
#include <driver/gpio.h>
#include <driver/rmt.h>
#include <esp_err.h>
#include <freertos/task.h>

const rmt_channel_t channel = RMT_CHANNEL_0;
const gpio_num_t pin = GPIO_NUM_2;
const rmt_item32_t data[] = {
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, },
	{ 32767, 1,	32767, 0, }
};
auto size = sizeof(data) / sizeof(data[0]);

extern "C" void app_main()
{

	const rmt_config_t rmt = {
		.rmt_mode = RMT_MODE_TX,
		.channel = channel,
		.clk_div = 255,
		.gpio_num = pin,
		.mem_block_num = 1,
		{
			.tx_config = rmt_tx_config_t{
				.loop_en = false,
				.carrier_freq_hz = 10000,
				.carrier_duty_percent = 0,
				.carrier_level = RMT_CARRIER_LEVEL_HIGH ,
				.carrier_en = false,
				.idle_level = RMT_IDLE_LEVEL_HIGH,
				.idle_output_en = true
			}
		}
	};

	ESP_ERROR_CHECK(rmt_config(&rmt));
	ESP_ERROR_CHECK(rmt_driver_install(channel, 0, 0));

	ESP_LOGI("rmt-test", "Will write first batch of items");
	rmt_write_items(channel, data, size, false);
	rmt_register_tx_end_callback([](const rmt_channel_t ch) {
		ESP_EARLY_LOGI("rmt-test", "Will write next batch of items");
		rmt_write_items(channel, data, size, false);
	});

	vTaskDelay(pdMS_TO_TICKS(3000));
	configASSERT(0);
}
```

The crash in the log is caused intentionally to show the premature `TX END`.

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[... 72 lines cut ...]
D (10) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (10) cpu_start: Starting scheduler on APP CPU.
D (332) heap_init: New heap initialised at 0x3ffe0440
D (332) heap_init: New heap initialised at 0x3ffe4350
D (342) rmt: Rmt Tx Channel 0|Gpio 2|Sclk_Hz 80000000|Div 255|Carrier_Hz 10000|Duty 0
V (352) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (352) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
V (362) rmt: RMT INTR : TX END                           🢀🢀🢀 this is premature
D (362) intr_alloc: Connected src 47 to int 13 (cpu 0)
I (372) rmt-test: Will write first batch of items        🢀🢀🢀 because it is first written here
V (2462) rmt: RMT INTR : TX END                          🢀🢀🢀 this is expected
I (2462) rmt-test: Will write next batch of items
.../esp32/main/main.cpp:287 (app_main)- assert failed!
abort() was called at PC 0x400d4c73 on core 0
0x400d4c73: app_main at .../esp32/main/main.cpp:287

Backtrace: 0x40089948:0x3ffc65e0 0x40089a6f:0x3ffc6600 0x400d4c73:0x3ffc6620 0x400d0b1f:0x3ffc6670
0x40089948: invoke_abort at .../esp32/esp-idf/components/esp32/panic.c:553

0x40089a6f: abort at .../esp32/esp-idf/components/esp32/panic.c:553

0x400d4c73: app_main at .../esp32/main/main.cpp:287

0x400d0b1f: main_task at .../esp32/esp-idf/components/esp32/cpu_start.c:429
```

koobest, 2018-01-04:
Hi, @chanibal,
Would you kindly tell me  which version of IDF are you using?
From the log I can see  you have the psram enabled，but gpio2 is used to access psram. the crash  may caused by this. you can try to use other gpio such as  gpio18,19,21...

negativekelvin, 2018-01-04:
@koobest gpio2 is used for sdcard not psram

koobest, 2018-01-04:
@negativekelvin, Will this issue occur if the interrupt flag is cleared in driver_install function?

chanibal, 2018-01-04:
@koobest: that was using ESP-IDF v3.0-dev-1057-g4048db35, as stated in the comment just under the issue. Both in the original issue and in the later reproduction.

koobest, 2018-01-09:
Hi, @chanibal,
The cause has been found, as you said. the tx_done flag is seted at power-on since there is really no data need to send. the solution is to clear the tx_done flag when initialization(in  rmt_config ).
Thanks for your feedback on this issue，we will fix  this issue in later version.

koobest, 2018-01-09:
@chanibal, this has a certain relationship with the `rmt_register_tx_end_callback`, if there is no callback, this problem will not happen, if you increase the delay time befor  `configASSERT`, you will see "RMT INTR : TX END"  printed.

FayeY, 2018-01-11:
Hi @chanibal, did the above suggestion help you resolve the issue?

koobest, 2018-01-22:
Hi, @chanibal, has your problem been solved? if not, I think you can update your IDF to the latest version.

FayeY, 2018-01-22:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-01-22 by FayeY

chanibal, 2018-01-22:
Sorry for the late reply. I have found another way around the issue before this, but still it's nice to hear that this issue in RMT will be fixed.

@koobest, regarding the suggestion from 2018-01-09 (increase delay so `TX END` will be printed 3 times): I don't understand the significance, this effect was intentional. I needed to write a RMT sequence and force a reboot (`configAssert(0)`) in the progress to show the bug. By giving a larger delay, you have just added one more cycle to the RMT.

As with the relation with `rmt_register_tx_end_callback`, then perhaps the callback is executed in wrong order? The question is if
```c++
	// ...
	ESP_LOGI("rmt-test", "Will write first batch of items");
	rmt_write_items(channel, data, size, false);
	configASSERT(0);
}
```
will have the same effect with the premature `TX END`.