espressif/esp-idf #1223: IEEE 1588 Support

ajvpot, 2017-11-03:
Hi, 

The datasheet lists the ESP32 as having hardware IEEE 1588 support, but the only reference I can find in the SDK to it is a "TODO" line. Is support for this feature implemented yet?

https://github.com/espressif/esp-idf/blob/master/components/ethernet/emac_main.c#L771

Thanks,
Alex

FayeY, 2017-11-10:
Hi, sorry that there is no update, it is still in our TODO list now. But we will let you know when there is any update.

ProfFan, 2018-02-04:
Hi,

We need this feature as well. Can you alternatively provide instructions and references so we can implement this ourselves?

limaconaudio, 2018-02-26:
Would there be a particular PHY required for this? What else is hidden away in the MAC?  For AVB devices you need IEEE 802.1AS :-)

sudo-ashish, 2018-06-25:
Hi,
Is the support for hardware ptp feature ready yet ?
Could not find any documentation or software support in official docs released till now.

rollet, 2019-02-19:
Is there any news about PTP status in ESP32 IDF/manuals ? It looks like frame timestamping is available , but there is no indication about clock control, PTP clock triggered interrupt, etc... 
Thanks

Grtschnk, 2019-04-30:
I am also very interested in this. :)
 
I wonder if it would be even possible to combine this with the ESP Mesh functionality (only very briefly looked into it), and as a result, have a swarm of almost-perfectly synced ESP32s.

rikkertkoppes, 2019-05-15:
Very interested also

Any pointers to an implementation would be greatly appreciated

ajvpot, 2019-05-16:
@Grtschnk Unfortunately, I don't think that would work. IRR1588 is an Ethernet feature. You'd need to be using a wired connection with a MAC.

You may have some luck with https://gitlab.com/painlessMesh/painlessMesh

peterjeremy, 2019-11-23:
The ESP32 TRM shows that the Ethernet MAC supports IEEE1588-2008.  IEEE802.1AS includes support for PTP over WiFi but AFAIK, the ESP32 doesn't support that.

brianbienvenu, 2020-03-06:
Any updates on this? Functional 1588v2 PTP support (ie, more than datasheet support) on the ESP32 would enable some interesting applications.

PureBasic, 2020-05-17:
Bump !
It would be great to get support for the IEEE1588 hardware or some documentation at least !

chuledeco, 2020-06-26:
I'm interested on this too!!!

chuledeco, 2020-06-26:
Does anyone found the way to implement PTP on ESP32?

callumbrieske, 2020-08-03:
Is there any further news on this?
The documentation for these features seems to be missing some information.

Where is the 'timestamp' DMA'd from? IE What is the source timer?
How do you enable the "Advanced Timestamping" features? I cant get the timestamps for either TX of RX frames, and the tech ref manual is very vague on the specifics of enabling these features.

Do these features actually exist as advertised, or have they never actually worked?

tionebrr, 2020-08-03:
@callumbrieske I didn't tried to go further on this. 
I think the 1588 timestamping may be a bit hard to make on an MCU and is not interesting for enough people. So the manufacturers just drop it. STMicro also had advertized PTP support for their Cortex MCUs, but have now removed the documentation for it. It seems like NXP are still supporting 1588 with code examples.
I think I'll just use an ethernet PHY with 1588 support (from Microchip). It might be more reliable on the long run as I can write my lib for that phy (or switch) and port it on any target.

doragasu, 2020-11-18:
Any news? Even if difficult, if doable, I am also interested. Maybe if we could have documentation about registers, we could implement it ourselves.

tionebrr, 2020-11-18:
I didn't get any further on that.

TooDissing, 2022-05-11:
Any updates on this? No real update since 2017...

ankurrajw, 2022-07-14:
I am also interested for this feature and hope it would be added to the ESP-IDF in the later releases.

ElliottSeer, 2023-01-19:
ESP32-C6 有希望实现这个功能吗

rollet, 2023-01-19:
> ESP32-C6 有希望实现这个功能吗

No Ethernet interface on ESP32-C6 from my knowledge...

callumbrieske, 2023-01-19:
Hopefully the ESP32-P4 will have working 1588.

jack0c, 2023-04-12:
ESP32-P4 will support PTP.

doragasu, 2023-04-12:
That's great news, thanks for commenting!

kostaond, 2025-04-14:
ESP32P4 supports HW time stamping and PTP example is available at https://github.com/espressif/esp-idf/tree/master/examples/ethernet/ptp

closed 2025-04-14 by kostaond