espressif/esp-idf #1225: Homekit compatibility

ToXiiK, 2017-11-04:
Hello,

I heard about homekit sdk in c of expressif , does anyone know how to get this sdk as well as the documentation? because I can not find any information on this sdk on google.

thank you in advance

Cordially

negativekelvin, 2017-11-04:
If you are mFi approved, contact espressif

ToXiiK, 2017-11-04:
i am not mfi but as homekit has been open to makers for noncommercial applications, sdk should be available for all?

btashton, 2017-11-05:
@negativekelvin I have the appropriate paperwork with Apple for MFI development, who in espressif should I reach out to about this?

negativekelvin, 2017-11-05:
@btashton would suggest you go through your sales contact at sales@espressif.com and ask them to have appropriate person contact you regarding access to Apple homekit software

igrr, 2017-11-08:
Yes, please contact sales for Homekit SDK related questions.

closed 2017-11-08 by igrr