espressif/esp-idf #1226: smartconfig

diegojpalaquibay, 2017-11-04:
Hello 
I cloned the master branch and I tried to use the example of smart config , and the esp32 chip can not connect with the router , i tried with different routers but it still cannot connect, then I changed to the branch release/v2.1 and the example works. I am trying to use the master branch because of  it has  improvements for the bluetooth, and the wifi bluetooth coexistence... Do u have any suggestion to solve the problem of the smartconfig ?
Thx for your help

Regards
Diego

yake827, 2017-11-09:
HI @diegojpalaquibay ,Could you please give a bit more details? I tested the smartconfig using the latest IDF and find it is OK.

laszlopataki, 2017-11-09:
Hello,

I had problems with the smartconfig too. 
It stopped after the below output and not stepped in SC_STATUS_LINK_OVER state.
```
...
I (239932) sc: SSID:IOT_DEMO_TEST
I (239932) sc: PASSWORD:123456789
```

After updating the IDF, It works fine now.
I hope it will help.

Regards,
Laszlo

azogue, 2017-11-09:
Hi, I'm also having a big trouble with smartconfig. It **works perfectly with v2.1**, but in **master** I'm getting a _guru meditation error_ every time. Was wondering about my code, but I've just copied the example and it crashes too.

This is the output I'm obtaining with the example:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4456
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:12980
entry 0x40078d84
I (375) wifi: wifi firmware version: 731ced4
I (376) wifi: config NVS flash: enabled
I (376) wifi: config nano formating: disabled
I (439) wifi: Init dynamic tx buffer num: 32
I (439) wifi: Init data frame dynamic rx buffer num: 32
I (440) wifi: Init management frame dynamic rx buffer num: 32
I (443) wifi: wifi driver task: 3ffc0114, prio:23, stack:4096
I (448) wifi: Init static rx buffer num: 10
I (452) wifi: Init dynamic rx buffer num: 32
I (456) wifi: Init rx ampdu len mblock:7
I (460) wifi: Init lldesc rx ampdu entry mblock:4
I (464) wifi: wifi power manager task: 0x3ffc5598 prio: 21 stack: 2560
I (508) wifi: mode : sta (24:0a:c4:11:a3:ac)
I (515) smartconfig: SC version: V2.6.8
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4011bc91  PS      : 0x00060930  A0      : 0x800dde8d  A1      : 0x3ffbff30
0x4011bc91: sc_wifi_scan_done at ??:?

A2      : 0x00000000  A3      : 0x3ffb9f4c  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x0000000a  A7      : 0x3ffaf930  A8      : 0x8011bc69  A9      : 0x3ffbff20
A10     : 0x3ffb9f4c  A11     : 0x3ffb1b96  A12     : 0x00000006  A13     : 0x3ffb9f52
A14     : 0x00001313  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x4011bc91:0x3ffbff30 0x400dde8a:0x3ffbff70 0x400de068:0x3ffbffc0 0x400de07b:0x3ffc0000 0x400dfc8b:0x3ffc0020 0x400dfd6e:0x3ffc0040 0x400e921b:0x3ffc0060
0x4011bc91: sc_wifi_scan_done at ??:?

0x400dde8a: free_bss_info at ??:?

0x400de068: free_bss_info at ??:?

0x400de07b: scan_enter_oper_channel_process at ??:?

0x400dfc8b: wifi_station_stop at ??:?

0x400dfd6e: ieee80211_timer_do_process at ??:?

0x400e921b: ppTask at ??:?
```

I'm using different esp32 boards, this is happening in all of them (now I'm with a WROVER KIT).

azogue, 2017-11-09:
My bad, it seems I wasn't entirely on _master_... Now the example is working for me. I see the smartconfig version has increased to 2.7.0. Let's hope it will work now

closed 2017-11-10 by FayeY

yrh79, 2017-11-27:
In certain version of the esp-idf (recent master branch), the re-connection of WIFI after smart config gets the SSID and password would get stuck (like what @laszlopataki saw) due to some stuck data in the wifi_config data structure. A totally new-initialized wifi_config_t data structure would have to be used to start the new connection, as below:

```
        case SC_STATUS_LINK:
            ESP_LOGI(TAG, "SC_STATUS_LINK");
            wifi_config_t *wifi_config = pdata;
            wifi_config_t new_wifi_config = { 0 }; <================
            ESP_LOGI(TAG, "SSID:%s", wifi_config->sta.ssid);
            ESP_LOGI(TAG, "PASSWORD:%s", wifi_config->sta.password);
            ESP_LOGI(TAG, "Disconnecting WIFI for reconnecting to AP");
            ESP_ERROR_CHECK( esp_wifi_disconnect() );
            ESP_LOGI(TAG, "setting WIFI configuration");

            // now fill up the new config struct 
            // (copy the ssid and password from old structure) <==== and below
            strlcpy((char*) new_wifi_config.sta.ssid, (char *) wifi_config->sta.ssid, sizeof(new_wifi_config.sta.ssid));
            if (wifi_config->sta.password) {
              strncpy((char*) new_wifi_config.sta.password, (char *) wifi_config->sta.password, sizeof(new_wifi_config.sta.password));
            }

            ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &new_wifi_config) );
            ESP_LOGI(TAG, "connecting WIFI");
            ESP_ERROR_CHECK( esp_wifi_connect() );
            break;
```

jackyc58, 2019-05-06:
I also have the same problem with ESP32-S, but it is normal on ESP32-WROVER, not the difference in RAM size.

Later, I added the wifi antenna and everything was fine.

capantojar3, 2019-05-30:
Hi, I have the same problem with a Wesp32-wrover module: the same code working fine in a ESP32-WROOM-32, any ideas?

I (212) smartconfig: SC version: V2.7.3
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4011bc0e  PS      : 0x00060b30  A0      : 0x801050e8  A1      : 0x3ffd1720  
0x4011bc0e: sc_wifi_scan_done at ??:?

A2      : 0x3ffd65c4  A3      : 0x00000001  A4      : 0x3ffb19e4  A5      : 0x3ffb75a0  
A6      : 0x3ffcaedc  A7      : 0x00000000  A8      : 0x8011bb9c  A9      : 0x00000000  
A10     : 0x3ffd65c4  A11     : 0x3ffb1862  A12     : 0x00000006  A13     : 0x3ffd65ca  
A14     : 0x00001313  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4011bc0e:0x3ffd1720 0x401050e5:0x3ffd1770 0x40105354:0x3ffd17c0 0x4010536b:0x3ffd1800 0x40107b34:0x3ffd1820 0x40107c5e:0x3ffd1840 0x40086e88:0x3ffd1860
0x4011bc0e: sc_wifi_scan_done at ??:?

0x401050e5: free_bss_info at ??:?

0x40105354: free_bss_info at ??:?

0x4010536b: scan_enter_oper_channel_process at ??:?

0x40107b34: wifi_station_stop at ??:?

0x40107c5e: ieee80211_timer_do_process at ??:?

0x40086e88: ppTask at ??:?

Rebooting...