espressif/esp-idf #1228: Flashing Error ESP32 (esptool)

kie-William, 2017-11-04:
I cant flash my esp when I run `make flash`
it shows that error , I have tried to config the serial port to` /dev/tty0 ` and `/dev/ttys0`
and it allows show an error like that 

> Flashing binaries to serial port /dev/ttyS0 (app at offset 0x10000)...
esptool.py v2.1
Traceback (most recent call last):
 File "/home/kirolos/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2524, in <module>
    _main()
  File "/home/kirolos/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2517, in _main
    main()
  File "/home/kirolos/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2246, in main
    esp = chip_class(args.port, initial_baud)
  File "/home/kirolos/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 177, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python2.7/dist-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 251, in open
    self._reconfigure_port(force_update=True)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 293, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'Input/output error')
/home/kirolos/esp/esp-idf/components/esptool_py/Makefile.projbuild:54: recipe for target 'flash' failed
make: *** [flash] Error 1

negativekelvin, 2017-11-05:
https://github.com/espressif/esp-idf/blob/3a271a4ae7df8a9049fbbb801feafca5043c31eb/docs/get-started/establish-serial-connection.rst#adding-user-to-dialout-on-linux

kie-William, 2017-11-05:
the just wasn't road by Ubuntu, I pressed the reset and the boot buttons then ubuntu saw it

closed 2017-11-05 by kie-William