espressif/esp-idf #1230: OTA update: invalid segment length

Robbiedobbie, 2017-11-06:
I'm trying to get the OTA update working on my esp32, however it's not accepting the firmware image as a valid one.

I compiled my firmware with make app, uploaded the resulting bin file to the chip through the serial flasher, which works fine.

If I then upload the very same image through http to the ESP, and use the esp_ota_begin/esp_ota_write/esp_ota_end functions to write the firmware, it complains about an invalid segment length when using the end function.

First I thought there was a bug in my http code resulting in some corrupt data to be passed onto the chip, but I checked a hexdump of the actual received buffers and the original file, and they are identical. If I check a hexdump of the target partition, some values in the header appear to be changed, and I'm not certain whether this is expected behaviour, or is the cause of my issues.

When booting the firmware, the bootloader gives the following:

```
I (92) boot: Partition Table:
I (96) boot: ## Label            Usage          Type ST Offset   Length
D (103) boot: load partition table entry 0x3f408000
D (108) boot: type=1 subtype=2
I (111) boot:  0 nvs              WiFi data        01 02 00009000 00005000
D (119) boot: load partition table entry 0x3f408020
D (124) boot: type=1 subtype=0
I (127) boot:  1 otadata          OTA data         01 00 0000e000 00002000
D (134) boot: load partition table entry 0x3f408040
D (139) boot: type=0 subtype=10
I (142) boot:  2 ota_0            OTA app          00 10 00010000 00100000
D (150) boot: load partition table entry 0x3f408060
D (155) boot: type=0 subtype=11
I (158) boot:  3 ota_1            OTA app          00 11 00110000 00100000
I (166) boot: End of partition table
D (170) boot: OTA data offset 0xe000
D (174) bootloader_flash: mmu set paddr=00000000 count=1
D (179) boot: OTA sequence values A 0xffffffff B 0xffffffff
D (185) boot: OTA sequence numbers both empty (all-0xFF)
I (190) boot: No factory image, trying OTA 0
D (195) boot: Trying partition index 0 offs 0x10000 size 0x100000
D (201) esp_image: reading image header @ 0x10000
D (206) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (213) esp_image: image header: 0xe9 0x07 0x02 0x02 40081218
V (219) esp_image: loading segment header 0 at offset 0x10018
V (224) esp_image: segment data length 0x1583c data starts 0x10020
V (231) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f400020
I (239) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1583c ( 88124) map
D (248) bootloader_flash: mmu set paddr=00010000 count=2
V (285) esp_image: loading segment header 1 at offset 0x2585c
D (285) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (286) esp_image: segment data length 0x3508 data starts 0x25864
V (292) esp_image: segment 1 map_segment 0 segment_data_offs 0x25864 load_addr 0x3ffb0000
I (301) esp_image: segment 1: paddr=0x00025864 vaddr=0x3ffb0000 size=0x03508 ( 13576) load
D (310) bootloader_flash: mmu set paddr=00020000 count=1
V (320) esp_image: loading segment header 2 at offset 0x28d6c
D (321) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (328) esp_image: segment data length 0x400 data starts 0x28d74
V (334) esp_image: segment 2 map_segment 0 segment_data_offs 0x28d74 load_addr 0x40080000
I (342) esp_image: segment 2: paddr=0x00028d74 vaddr=0x40080000 size=0x00400 (  1024) load
D (351) bootloader_flash: mmu set paddr=00020000 count=1
V (357) esp_image: loading segment header 3 at offset 0x29174
D (362) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (369) esp_image: segment data length 0x6e94 data starts 0x2917c
V (375) esp_image: segment 3 map_segment 0 segment_data_offs 0x2917c load_addr
D (392) bootloader_flash: mmu set paddr=00020000 count=2
V (410) esp_image: loading segment header 4 at offset 0x30010
D (410) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
V (413) esp_image: segment data length 0x77424 data starts 0x30018
V (419) esp_image: segment 4 map_segment 1 segment_data_offs 0x30018 load_addr 
I (427) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x77424 (488484) map
D (436) bootloader_flash: mmu set paddr=00030000 count=8
V (617) esp_image: loading segment header 5 at offset 0xa743c
D (617) bootloader_flash: mmu set block paddr=0x000a0000 (was 0xffffffff)
V (618) esp_image: segment data length 0x97d0 data starts 0xa7444
V (624) esp_image: segment 5 map_segment 0 segment_data_offs 0xa7444 load_addr 
I (633) esp_image: segment 5: paddr=0x000a7444 vaddr=0x40087294 size=0x097d0 ( 38864) load
D (641) bootloader_flash: mmu set paddr=000a0000 count=2
V (663) esp_image: loading segment header 6 at offset 0xb0c14
D (663) bootloader_flash: mmu set block paddr=0x000b0000 (was 0xffffffff)
V (664) esp_image: segment data length 0x0 data starts 0xb0c1c
V (670) esp_image: segment 6 map_segment 0 segment_data_offs 0xb0c1c load_addr 0x400c0000
I (679) esp_image: segment 6: paddr=0x000b0c1c vaddr=0x400c0000 size=0x00000 (     0) load
D (688) bootloader_flash: mmu set paddr=000b0000 count=1
V (693) esp_image: image start 0x00010000 end of last section 0x000b0c1c
D (700) bootloader_flash: mmu set block paddr=0x000b0000 (was 0xffffffff)
D (707) esp_image: Calculated hash: e9b3c2da
D (711) bootloader_flash: mmu set paddr=000b0000 count=1
I (726) boot: Loaded app from partition at offset 0x10000
```

However, when checking the file after upload, esp_image shows the following:

```
D (206636) esp_image: reading image header @ 0x110000
D (206636) esp_image: image header: 0xe9 0x07 0x02 0x02 40081218
V (206636) esp_image: loading segment header 0 at offset 0x110018
V (206636) esp_image: segment data length 0x1583c data starts 0x110020
V (206646) esp_image: segment 0 map_segment 1 segment_data_offs 0x110020 load_addr 0x3f400020
I (206656) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x1583c ( 88124) map
V (206726) esp_image: loading segment header 1 at offset 0x12585c
V (206726) esp_image: segment data length 0x64252820 data starts 0x125864
E (206736) esp_image: invalid segment length 0x64252820
```

Is there something obvious I'm missing here?

igrr, 2017-11-08:
If the data in flash does not match the contents of the buffer you received over HTTP, that is definitely an issue.

We have a work-in-progress change which adds optional verification to all SPI flash write operations. I hope it will land on github next week.

In the meantime, could you please compare the data written to flash with the expected data to see if
a) data corruption happens the same way on each upload (same address, same value)
b) is there any specific pattern of the data corruption?

You can use `esptool.py read_flash` operation to read the contents of the partition and do the comparison on the PC.

Robbiedobbie, 2017-11-08:
After trying to find a pattern I started seeing a correlation with filesize. Turns out that even though the binary arrived fine in my http handling code, somewhere I was corrupting memory causing some wrong data being written to chip...

So in the end it turned out it was a bug in my code after all... Sorry for the inconvenience.

closed 2017-11-08 by Robbiedobbie

laukik-hase, 2022-02-17:
Hello, @zark-zeugan!

As suggested by other users in the follow-up comments to your issue ([here](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/132#issuecomment-1039847451)), this is mostly related to power issues on your board. Did those suggestions help?

On another note, could you try reading the flash after receiving the OTA update and comparing the binaries? You can use [`esptool.py`](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#read-flash-contents-read-flash) for doing that.