espressif/esp-idf #1231: UART send break without any data

asuivelentine, 2017-11-06:
Hi,

I need to set up a UART connection for my project. The receiver waits for a specific 'Start Communication' pattern which looks like this:
```
//Tx Signal: --------------                 -------------------
//                         |                |
//                         |____0.025sec____|
```

In recent projects the pull down is done by sending break. 
From specs the only way to send breaks is to use the uart_write_bytes_with_break function. 
```
int uart_write_bytes_with_break(uart_port_t uart_num, const char *src, size_t size, int brk_len)
``` 

How is it possible to only send that pattern? It's important that the pull down time is exactly 0.025 sec.
thanks.

chanibal, 2017-11-06:
Maybe don't use UART at all. RMT is great for this kind of exact timing things. You'll have to implement UART yourself, but it's very easy. You will be constrained to the message length, though. 

If you can downgrade "exactly" to something around +/- 10ms, then a task or timer with `uart_write_bytes` will do too. 

Also, why doesn't `uart_write_bytes_with_break` work for you?

asuivelentine, 2017-11-06:
Hi, 
thanks for your response. 
It has to be UART. It's defined by K-Line Spec (10400 Baud, single wire interface)

Exactly means +/- 1ms. So `uart_write_bytes` won't be exact enough.

I cant use `uart_write_bytes_with_break` because it will pull up the line between data and break
```
-----            ---    --------
     |          |    |  |
      ----------      --
       Data          break
```

So the only possibility is to create my own UART driver?!

chanibal, 2017-11-06:
As far as I see UART always sends zeros in the break: https://github.com/espressif/esp-idf/blob/ff6a3b1a11032798565d7b78319444aa81881542/components/soc/esp32/include/soc/uart_reg.h#L738

But what is a zero is configurable: https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/uart.h#L259

Maybe you could attach to some UART interrupts and be able to set up these things on the fly?

Not sure if it's the only way, someone could have a better idea. Especially someone from Espressif.

negativekelvin, 2017-11-06:
https://esp32.com/viewtopic.php?f=13&t=1476

asuivelentine, 2017-11-06:
`uart_set_line_inverse` looks good. I can create a nice square wave signal.
With a little fine-tuning regarding the timing it should work.
Thank you for that idea.

closed 2017-11-06 by asuivelentine