espressif/esp-idf #1232: esp hanging

ikawaljeet, 2017-11-06:
So normally when esp is rebooted by external reset or esp_restart, logs are like this:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:5668
load:0x40078000,len:0
load:0x40078000,len:12188
entry 0x40078f5c

After this it proceeded normal 

Although it is not reproducible every time, once i got this:
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:3488
load:0x40078000,len:7636
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034  

After this it hanged

ikawaljeet, 2017-11-06:
After calling esp_restart() , this log was coming, after that, it was not able to restart
I (72184) wifi: state: run -> init (0)
I (72184) wifi: pm stop, total sleep time: 0/25971084

I (72184) wifi: n:10 0, o:10 0, ap:255 255, sta:10 0, prof:1
I (72234) wifi: flush txq
I (72234) wifi: stop sw txq
I (72234) wifi: lmac stop hw txq

negativekelvin, 2017-11-06:
Have you updated with this commit 
https://github.com/espressif/esp-idf/commit/abacf8d2a0347a4bf95315cdb227e9b8b1c93794

ikawaljeet, 2017-11-07:
So i updated the code with 1613044 commit today, which is ahead of abacf8d, it worked better still some issue is there.

I rebooted 8 esp32 over more than 250 times through esp_restart() , one got stopped after 325 reboots and one after 285 reboots. Earlier this was happening around 100 reboots. 

I was not able to see the logs of those esp32 which stopped since i have only one uart bridge. Also, I am using uart with flow control.

igrr, 2017-11-08:
@ikawaljeet thanks for reporting this. Could you please give a bit more details? Which development boards do you use, how are they powered, what options are selected for CPU and RTC SLOW_CLK frequencies in Kconfig, whether you have enabled dynamic frequency scaling, whether there are some peripherals attached to the ESP32?

We have been doing similar test today with a combination of Devkit-C and Devkit-J boards, and haven't found issues so far. Our test probably differs from yours in certain way, so we need to find what this difference is.

igrr, 2017-12-08:
Closing as can't reproduce. Please reopen if the issue persists and you can supply more details.

closed 2017-12-08 by igrr