espressif/esp-idf #1233: VSPI Bus - [SCK, MOSI] on [IO34, IO35] not working

neilpanchal, 2017-11-06:
Hello,

I have been rigorously testing the pin mux abstraction provided by ESP-IDF framework. Documentation describes in several places that any GPIO pin can be mapped for VSPI/HSPI busses.

I mapped SCK & MOSI pins to bunch of different ones by modifying the `spiAttachMOSI()` and `spiAttachSCK()` functions in `esp32-hal-spi.c` file. All pins work except for IO34 and IO35. Are these reserved for some other purpose? 

I am using ESP32-Devkit-C module as well as tried it on ESP32-WROVER-KIT V2. Both exhibit the same problem.

Spritetm, 2017-11-06:
GPIO 34 and 35 do not have output drivers, so can't be used as outputs. This goes for all GPIOs in the 34-39 range.

neilpanchal, 2017-11-06:
@Spritetm Thanks for clarifying. Is this documented somewhere in ESP-IDF docs or the datasheet? If not, we should.

negativekelvin, 2017-11-06:
@neilpanchal datasheet section 2.2 pin description type column

neilpanchal, 2017-11-06:
@negativekelvin Thank you. I missed this :)

closed 2017-11-06 by neilpanchal