espressif/esp-idf #1234: GPIO Interrupt sometimes doesn't fire?

chanibal, 2017-11-06:
I had issues with debouncing a simple button, I traced them down to a ISR registered by `gpio_isr_handler` are not always executing. A simple test case:

```c++
const gpio_num_t pin = GPIO_NUM_34;
volatile int rise = 0;
volatile int fall = 0;

extern "C" void IRAM_ATTR triggered_isr(void* instance) {
	if (gpio_get_level(pin)) rise++; 
	else fall++;
}

extern "C" void app_main() {
	gpio_config_t config = {
		.pin_bit_mask = 1ull << pin,
		.mode = GPIO_MODE_INPUT,
		.pull_up_en = GPIO_PULLUP_DISABLE,
		.pull_down_en = GPIO_PULLDOWN_DISABLE,
		.intr_type = GPIO_INTR_ANYEDGE
	};
	ESP_ERROR_CHECK(gpio_config(&config));

	gpio_install_isr_service(0);
	gpio_isr_handler_add(pin, triggered_isr, nullptr);

	for (;;) {
		vTaskDelay(pdMS_TO_TICKS(1000));
		ESP_LOGI("button-test", "rise=%04d, fall=%04d, diff = %04d", rise, fall, (rise-fall));
	}
}
```

Will yield results like:
```
I (10640) button-test: rise=0000, fall=0000, diff = 0000
I (11640) button-test: rise=0000, fall=0000, diff = 0000
I (12640) button-test: rise=0000, fall=0000, diff = 0000
I (13640) button-test: rise=0000, fall=0000, diff = 0000
I (14640) button-test: rise=0000, fall=0000, diff = 0000
I (15640) button-test: rise=0000, fall=0000, diff = 0000
I (16640) button-test: rise=0000, fall=0000, diff = 0000
I (17640) button-test: rise=0000, fall=0000, diff = 0000
I (18640) button-test: rise=0001, fall=0002, diff = -001
I (19640) button-test: rise=0002, fall=0003, diff = -001
I (20640) button-test: rise=0004, fall=0004, diff = 0000
I (21640) button-test: rise=0005, fall=0006, diff = -001
I (22640) button-test: rise=0007, fall=0007, diff = 0000
I (23640) button-test: rise=0008, fall=0009, diff = -001
I (24640) button-test: rise=0010, fall=0011, diff = -001
I (25640) button-test: rise=0014, fall=0014, diff = 0000
I (26640) button-test: rise=0017, fall=0019, diff = -002
I (27640) button-test: rise=0020, fall=0021, diff = -001
I (28640) button-test: rise=0028, fall=0029, diff = -001
I (29640) button-test: rise=0034, fall=0037, diff = -003
I (30640) button-test: rise=0040, fall=0043, diff = -003
I (31640) button-test: rise=0043, fall=0046, diff = -003
I (32640) button-test: rise=0045, fall=0049, diff = -004
I (33640) button-test: rise=0049, fall=0052, diff = -003
```

Am I using interrupts correctly? Is one interrupt hiding another that is quick in succession? If so, how can I prevent this?

I have tried `std::atomic<int>` but it did not help.

negativekelvin, 2017-11-06:
Obviously multiple interrupts can happen during your vtaskdelay

negativekelvin, 2017-11-06:
Add hw filter or use level interrupt

chanibal, 2017-11-06:
If by hw filter, you mean a capacitor then this happens even more with one.

I updated the solution to use level interrupts, to be exact:
```C++
extern "C" void IRAM_ATTR triggered_isr(void* instance) {
	if (gpio_get_level(pin)) {
		gpio_set_intr_type(pin, GPIO_INTR_LOW_LEVEL);
		rise++;
	}
	else {
		gpio_set_intr_type(pin, GPIO_INTR_HIGH_LEVEL);
		fall++;
	}
}
// ...
		.intr_type = GPIO_INTR_LOW_LEVEL
```

But this still fails (although, I need to try harder to break it):
```
I (1720) button-test: rise=0000, fall=0000, diff = 0000
I (2720) button-test: rise=0000, fall=0000, diff = 0000
I (3720) button-test: rise=0000, fall=0001, diff = -001
I (4720) button-test: rise=0001, fall=0001, diff = 0000
I (5720) button-test: rise=0001, fall=0001, diff = 0000
I (6720) button-test: rise=0003, fall=0003, diff = 0000
I (7720) button-test: rise=0008, fall=0008, diff = 0000
I (8720) button-test: rise=0010, fall=0011, diff = -001
I (9720) button-test: rise=0011, fall=0011, diff = 0000
I (10720) button-test: rise=0011, fall=0011, diff = 0000
I (11720) button-test: rise=0011, fall=0011, diff = 0000
I (12720) button-test: rise=0011, fall=0012, diff = -001
I (13720) button-test: rise=0012, fall=0012, diff = 0000
I (14720) button-test: rise=0012, fall=0012, diff = 0000
I (15720) button-test: rise=0013, fall=0014, diff = -001
I (16720) button-test: rise=0018, fall=0019, diff = -001
I (17720) button-test: rise=0023, fall=0025, diff = -002
I (18720) button-test: rise=0031, fall=0033, diff = -002
I (19720) button-test: rise=0039, fall=0040, diff = -001
I (20720) button-test: rise=0046, fall=0048, diff = -002
I (21720) button-test: rise=0053, fall=0055, diff = -002
I (22720) button-test: rise=0061, fall=0063, diff = -002
I (23720) button-test: rise=0066, fall=0072, diff = -006
I (24720) button-test: rise=0066, fall=0072, diff = -006
I (25720) button-test: rise=0066, fall=0072, diff = -006
I (26720) button-test: rise=0066, fall=0072, diff = -006
```

I guess that the issue is in one of:
* `gpio_get_level` - is there a better way to check what triggered the ISR?
* `gpio_set_intr_type` - is it OK to call this from ISR context?
* `rise++` - probably not atomic enough (again, tried with std::atomic, but did not help - does it even work on ESP32?)

chanibal, 2017-11-06:
I think I managed to fix it with replacing `gpio_get_level` with a flag, still not very elegant and I hope there is a better solution.

This is what helped:
```C++
volatile bool isHigh = false;
extern "C" void IRAM_ATTR triggered_isr(void* instance) {
	if (isHigh) {
		gpio_set_intr_type(pin, GPIO_INTR_LOW_LEVEL);
		rise++;
		isHigh=false;
	}
	else {
		gpio_set_intr_type(pin, GPIO_INTR_HIGH_LEVEL);
		fall++;
		isHigh=true;
	}
}
```

negativekelvin, 2017-11-06:
Yes that is good because it filters out any very short pulses but does not miss any real transitions and now you just need to measure the time between interrupts.

chanibal, 2017-11-07:
As to debounce timing, I have already filtered this out in "real" code, and it works properly now. Still, this doesn't look as elegant as it should. There has to be somewhere an value of what has triggered the current interrupt - something that will replace the `isHigh` variable.

Anyway, it works now, thanks for the help, closing.

closed 2017-11-07 by chanibal