espressif/esp-idf #1235: Can not make

lixianyu, 2017-11-07:
lxy@lxy-PC MINGW32 /c/myProject/esp32/Rhea1_TEST
$ make
make[1]: 进入目录“/c/360yunpan/360tong_bu/Project20160127/Dione/esp-idf3/tools/kconfig”
lxdialog/check-lxdialog.sh -check cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD -lncursesw -lintl
make[1]: *** 没有规则可制作目标“/usr/include/sys/_locale.h”，由“lxdialog/checklist.o” 需求。 停止。
make[1]: 离开目录“/c/360yunpan/360tong_bu/Project20160127/Dione/esp-idf3/tools/kconfig”
make: *** 没有规则可制作目标“/c/360yunpan/360tong_bu/Project20160127/Dione/esp-idf3/tools/kconfig/conf”，由“/c/myProject/esp32/Rhea1_TEST/build/include/config/auto.conf” 需求。 停止。

I use esp32_win32_msys2_environment_and_toolchain-20170918.zip toolchain I'm Win10 64bit.
Who can help me?

Gfast2, 2017-11-07:
Hi @lixianyu ,

Perhapse you wanna `make menuconfig`. save & exit it. which generate your `规则`. and then try it again. 

Good luck

lixianyu, 2017-11-08:
Hi，@Gfast2 ，谢谢。
重新clone一下，再编译就OK啦。^__^

closed 2017-11-08 by lixianyu