espressif/esp-idf #1236: Touch Sensor: initializing touchpad module interferes with ADC readings

bmetman, 2017-11-07:
Dear Espressif, 

I have been working with the Touch Sensor module as described [here](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/touch_pad.html). I'm executing the following code:

```
adc1_config_width(ADC_WIDTH_12Bit);
adc1_config_channel_atten(ADC1_CHANNEL_5, ADC_ATTEN_11db);
...
touch_pad_init();
for(;;){
  ESP_LOGI(TAG, "ADC: %d," adc1_get_raw(ADC1_CHANNEL_5));
  vTaskDelay(1000 / portTICK_PERIOD_MS);
}
```
Because in this case, ADC1_CHANNEL_5 is associated with touchpad 8, the ADC readings for ADC1_CHANNEL_5 are not accurate anymore and quite random. If the touch_pad_init() function is not executed, the ADC readings are fine. 

Is there an option to disable the touch sensor initialization for specific touchpads? Or is there another way to solve this? Right now whenever I want to read the ADC value of adc1 channel 5 I deinitialize the touchpad module, and after reading I initialize it again. This is very slow and not the preferred solution. 

By the way, if no touch filter is initialized and the touch sensor module is deinitialized, the following error message also occurs: 

`E (25085) RTC_MODULE: /home/user/esp/esp-idf/components/driver/./rtc_module.c:913 (touch_pad_filter_delete):Touch pad filter not initialized`

This is also incorrect. Thank you in advance for your help.

bmetman, 2017-11-21:
Hey there, any indication on when you have time look into this issue?

jmorsap, 2018-01-17:
I am also experiencing the same issue. When I initialize the touch sensor on gpio13, my adc measurements on gpio32 and gpio33 get very noisy. 

GPIO32 is defined as an input, and is set up for ADC. GPIO13 is set up for the touch sensor, but the ESP32 is driving the touch senor output to both pins. The image below is the signal on GPIO32, which is NOT set up as a touch sensor. 

[image]

negativekelvin, 2018-01-17:
Have you checked the value of SENS.sar_touch_enable.touch_pad_worken

jmorsap, 2018-01-18:
SENS.sar_touch_enable.touch_pad_worken returns a value of 1023 in my code. 

How is this mask/value used?

jmorsap, 2018-01-18:
This is defined in the sens_struct.h file, line 766. https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/sens_reg.h

negativekelvin, 2018-01-18:
Try setting it to 0 before you configure the touchpads.

jmorsap, 2018-01-18:
This works! I appreciate your help. 

Is this a bug? Or is there something I missed when setting up my touch sensor?

negativekelvin, 2018-01-18:
Bug

costaud, 2018-01-29:
We are working on this issue, I find something, will fix this soon. @bmetman

closed 2018-07-12 by FayeY