espressif/esp-idf #1241: Documentations lacks examples on how to read Hall sensor from ULP co-processor

baybal, 2017-11-07:
As I understand, before doing readings from pin 0 and 3, you have to set up Hall configuration registers in a certain way. Current register spec does not explain that in detail.

igrr, 2017-11-08:
Good point about adding such an example, we'll add it along with ULP I2C example.

In the meantime, I think TRM does have the information you are looking for: 

- Pins should be initialized same was as for ADC (see ulp_adc example),
- Hall sensor power on and phase should be controlled by ULP (`SENS_HALL_PHASE_FORCE=0`, `SENS_XPD_HALL_FORCE=0`),
- ADC should be controlled by ULP (`adc1_ulp_enable()`, same as in ulp_adc example). 
- After that you should be able to use ADC instruction of the ULP. Per TRM, Sel=0 (for ADC1), Sar_Mux = 9 returns the result for phase 1, Sar_Mux = 10 returns the result for phase 0.

TemcoHeng, 2017-12-13:
I make an example here: https://github.com/espressif/esp-iot-solution/tree/master/examples/ulp_examples/ulp_hall_sensor , FYI.

bvernoux, 2017-12-13:
Does it is possible to have such type of example (especially the Readme.md) in English and not Japanese/Chinese ... ?

TemcoHeng, 2018-01-17:
@bvernoux  English version doc will be updated soon.

TemcoHeng, 2018-03-08:
https://github.com/espressif/esp-iot-solution/blob/master/examples/ulp_examples/ulp_hall_sensor/readme_en.md, update english doc.

jack0c, 2018-03-21:
@FayeY Try to close this issue.

closed 2018-03-21 by FayeY