espressif/esp-idf #1242: BLE communication and sleep

pit001, 2017-11-08:
Hi !

I want to create a battery powered device with bluetooth communication. The device (client) should advertice and be connectable (or accept scan&responce) to receive new data.
Let's say a kind of SmartWatch with updates fom Smartphone. 

The Device is sleeping the most time, wakes up , advertices, wait a short time for data and goes sleep again. At the first time the BLE GAP/GATT have to be initialized, later it's connectable (if not already) and in the best case only data are received.

How can I receive new data by BLE and sleep between the updates without to begin every time a complete new init of BT and create a new connectioin if the ESP32 every time restarts completly? 

Is this possible ?

regards Pit

lucazader, 2017-11-08:
This is not currently possible with the ESP32.

Eventually this feature will be added to the idf (from what esp_iggr has said) but currently they way the bluetooth stack is configured it is not possible.

pit001, 2017-11-08:
Thx for te fast reply and the clear answer. As an alternative for now - is there an example using the scan&responce system (ADV_TYPE_SCAN_IND). 
So during wakeup do an adverticement, wait a short time for an connection from central (client). If no, go sleep. If yes, keep alife and respond (up to three times) and  after finish go sleep.

lucazader, 2017-11-08:
I dont think there are any examples in the IDF it's self. no.

pit001, 2017-11-08:
Thx. IMO more and more I come to the point that the ESP32 is useless with real BLE for the moment !

Weijian-Espressif, 2018-03-13:
sorry for taht, we will add ble sleep in v3.1.

Weijian-Espressif, 2018-05-21:
BLE sleep has merged to master, please update to the latest IDF and try.

closed 2018-05-21 by FayeY