espressif/esp-idf #1244: APLL doesn't work in DAC and PDM modes

MrBuddyCasino, 2017-11-08:
Is this supposed to work or not? If not, maybe a validation of the settings could be performed before starting the I2S driver?

OtherCrashOverride, 2017-11-09:
I also encountered this issue in a project not related to ESP32_MP3_Decoder.

HanVertegaal, 2018-06-15:
Using I2S in RX SLAVE mode, setting `i2s_config.use_apll = false` works, but `i2s_config.use_apll = true` does not. 

With apll=false I see the following line in my log: 

I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 32, CLKM: 13, BCK: 4, MCLK: 12292917.167, SCLK: 3076864.000000, diva: 64, divb: 1

With apll=true we get:

I2S: APLL: Req RATE: 48000, real rate: 22385.387, BITS: 32, CLKM: 1, BCK_M: 349, MCLK: 500000000.000, SCLK: 62500000.000000, diva: 1, divb: 0

In V3.0 this same code worked with appl=true.

L-KAYA, 2025-01-17:
APLL part has been refactored since v5.0. Will close this issue

closed 2025-01-17 by Alvin1Zhang