espressif/esp-idf #1247: ASSERT_ERR(0) in BT libbtdm_app.a during A2DP sink reception

jovermann, 2017-11-08:
I observed an assert while playing back (= receiving) A2DP data on the ESP32:

ASSERT_ERR(0), in lc_lmppdu.c at line 66
(this is apparently in components/bt/lib/libbtdm_app.a)

After this the watchdog triggers and the system is restarted (which works ok). But of course the BT connection is disconnected and the playback stops. 

The assert occurs randomly and only very rarely, after hours of streaming. 
The setup is:
- ESP32-DevKitC with rev1 silicon
- I2S DAC connected (and playing back)
- WiFi is disabled on the ESP32
- BT Classic is enabled on the ESP32, using BTstack (not Bluedroid) (roughly Oct 24).
- BT connection and pairing worked flawlessly for hours, A2DP playback is working.
- esp-idf at 2c95a77cf93781f296883d5dbafcdc18e4389656 (Oct 24)

What is asserted with this assert? Is this an invalid input to a BT function? Is this an out of memory condition? Is this an internal error?

Sorry, this is very hard to reproduce. I cannot even claim it is reproducible at all. I am merely hoping that a developer of libbtdm_app.a can look into the assert of gains information from this issue (e.g. that the asserted situation really occurs in real life).

I am already trying to reproduce this with the latest esp-idf/BTstack.

TianHao-Espressif, 2018-04-03:
@jovermann , the latest master branch have resolve the problem. It's due to the interrupt react slowly, cause bt codes cannot handle in time. Although the bug is fixed, please check your code, if any code lock interrupt(portENTER_CRITICAL) too long time.

Wth-Esp, 2020-05-14:
Hi all,

This issue has been fixed. Please pull the latest IDF.

Thanks
Weitianhua

closed 2020-06-11 by Alvin1Zhang