espressif/esp-idf #1250: Heap operation assert: "free() target pointer is outside heap areas"

Cooler1989, 2017-11-09:
Chip is ESP32D0WDQ6 (revision 1)

```
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "~/embedded/esp-idf/components/heap/./heap_caps.c", line 274, function: heap_caps_free
abort() was called at PC 0x400d4e4b on core 0

Backtrace: 0x40089320:0x3ffd0b30 0x4008941f:0x3ffd0b50 0x400d4e4b:0x3ffd0b70 0x400825df:0x3ffd0ba0 0x40082a21:0x3ffd0bc0 0x4000bec7:0x3ffd0be0 0x4012b14e:0x3ffd0c00 0x4012b1f4:0x3ffd0c30 0x4012b27a:0x3ffd0c50
```

Bisecting with git yields:
```
Bisecting: 0 revisions left to test after this (roughly 0 steps)                                                                    
[85eb5d4374cb47fd8a3dcfcc314743471b88b851] component/bt : fix bug of sw interrupt cause to run btdm to different cpu core.
```

Details:                                                                              
```
git bisect start    
# good: [062be56c3cac4accca0be9f2eeb5eaff60590f4f] docs: add power management API reference                                 
git bisect good 062be56c3cac4accca0be9f2eeb5eaff60590f4f                                                                    
# bad: [8e04b8615331583840312c140818e8ca5aa5ea2a] Merge branch 'bugfix/tg_wdt_reboot_uart_garbage' into 'master'            
git bisect bad 8e04b8615331583840312c140818e8ca5aa5ea2a                                                                     
# bad: [4048db35b6a1ac4216150480dcacdc4230f0baa2] Merge branch 'bugfix/unit_test_refactored_timer_api_usage' into 'master'  
git bisect bad 4048db35b6a1ac4216150480dcacdc4230f0baa2                                                                     
# good: [f76a3c45c02c322b7154d66430d802f0df5d7238] newlib: Disable sys/types.h implementations of FD_SET, etc.              
git bisect good f76a3c45c02c322b7154d66430d802f0df5d7238                                                                    
# good: [baf1641c297622d6f30278c8a864fd75defdbd6e] Merge branch 'bugfix/uart_set_default_idle_time_to_zero' into 'master'   
git bisect good baf1641c297622d6f30278c8a864fd75defdbd6e                                                                    
# good: [00bf160f94d177560fead6ba0dc4c9beef03df26] Merge branch 'bugfix/pthread_once' into 'master'                         
git bisect good 00bf160f94d177560fead6ba0dc4c9beef03df26                                                                    
# good: [ffeecde9e2c1617e26f51d0ab57f9752e2b9cd25] Merge branch 'feature/docs_update_timer_api' into 'master'               
git bisect good ffeecde9e2c1617e26f51d0ab57f9752e2b9cd25                                                                    
# bad: [a9b63b796e6af11ebbf53a81cce2b238071a7fdb] Merge branch 'bugfix/btdm_cpu_core' into 'master'                         
git bisect bad a9b63b796e6af11ebbf53a81cce2b238071a7fdb           
```

projectgus, 2017-11-09:
Thanks for the bug report and the bisect. What code are you running when you get this?

Can you please decode the backtrace addresses? You can do this with idf_monitor.py or by running addr2line manually (command also shown here):
http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html?highlight=monitor#automatically-decoding-addresses

Cooler1989, 2017-11-10:
I am new to the project so I do not know basic tools yet, but I will decode it later today ( within next 16 hours), but for now I can tell you that all the code I was running was taken from repository without any changes. I used bluetooth examples. This particular was from examples/bluetooth/gatt_server.
I hope that helps.

Cooler1989, 2017-11-10:
So I did run idf_monitor. This is the result:

```
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/cooler1989/embedded/esp-idf/components/heap/./heap_caps.c", line 274, function: heap_caps_free
abort() was called at PC 0x400d4e4b on core 0
0x400d4e4b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40089320:0x3ffd0b30 0x4008941f:0x3ffd0b50 0x400d4e4b:0x3ffd0b70 0x400825df:0x3ffd0ba0 0x40082a21:0x3ffd0bc0 0x4000bec7:0x3ffd0be0 0x4012b14e:0x3ffd0c00 0x4012b1f4:0x3ffd0c30 0x4012b27a:0x3ffd0c50
0x40089320: invoke_abort at /home/cooler1989/embedded/esp-idf/components/esp32/./panic.c:553

0x4008941f: abort at /home/cooler1989/embedded/esp-idf/components/esp32/./panic.c:553

0x400d4e4b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x400825df: heap_caps_free at /home/cooler1989/embedded/esp-idf/components/heap/./heap_caps.c:274 (discriminator 1)

0x40082a21: _free_r at /home/cooler1989/embedded/esp-idf/components/newlib/./syscalls.c:42

0x4012b14e: btdm_rom_ref_init at ??:?

0x4012b1f4: r_rw_pre_main at arch_main.c:?

0x4012b27a: btdm_controller_task at ??:?

Rebooting...

```
I hope it helps.

projectgus, 2017-11-10:
Thanks @Cooler1989 

@TianHao-Espressif PTAL :)

Oliv4945, 2017-11-12:
I have the same thing here after pulling esp-idf from #de750e99214a51c1bcfdb42b5ada931652a3c531 to  #8e04b8615331583840312c140818e8ca5aa5ea2a.
With default config on gatt_client demo example

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5832
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13804
entry 0x40079030
I (74) boot: Detected ESP32
I (34) boot: ESP-IDF v3.0-dev-1147-g8e04b861-dirty 2nd stage bootloader
I (34) boot: compile time 00:05:01
I (44) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (45) boot: SPI Mode       : DIO
I (49) boot: SPI Flash Size : 4MB
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2ba20 (178720) map
I (161) esp_image: segment 1: paddr=0x0003ba48 vaddr=0x3ffc0000 size=0x02a78 ( 10872) load
I (166) esp_image: segment 2: paddr=0x0003e4c8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (168) esp_image: segment 3: paddr=0x0003e8d0 vaddr=0x40080400 size=0x01740 (  5952) load
I (179) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x67f80 (425856) map
0x400d0018: _stext at ??:?

I (334) esp_image: segment 5: paddr=0x000a7fa0 vaddr=0x40081b40 size=0x10460 ( 66656) load
0x40081b40: timer_overflow_happened at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/esp32/./esp_timer_esp32.c:132
 (inlined by) timer_alarm_isr at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/esp32/./esp_timer_esp32.c:206

I (361) esp_image: segment 6: paddr=0x000b8408 vaddr=0x400c0000 size=0x00000 (     0) load
I (372) boot: Loaded app from partition at offset 0x10000
I (372) boot: Disabling RNG early entropy source...
I (373) cpu_start: Pro cpu up.
I (377) cpu_start: Starting app cpu, entry point is 0x40080eec
0x40080eec: call_start_cpu1 at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/esp32/./cpu_start.c:222

I (366) cpu_start: App cpu up.
I (387) heap_init: Initializing. RAM available for dynamic allocation:
I (394) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (400) heap_init: At 3FFCA920 len 000156E0 (85 KiB): DRAM
I (406) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (413) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (419) heap_init: At 40091FA0 len 0000E060 (56 KiB): IRAM
I (425) cpu_start: Pro cpu start user code
I (108) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/heap/./heap_caps.c", line 274, function: heap_caps_free
abort() was called at PC 0x400d4b83 on core 0
0x400d4b83: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x400894f0:0x3ffd0fc0 0x400895ef:0x3ffd0fe0 0x400d4b83:0x3ffd1000 0x40082623:0x3ffd1030 0x40082a65:0x3ffd1050 0x4000bec7:0x3ffd1070 0x4012b976:0x3ffd1090 0x4012ba1c:0x3ffd10c0 0x4012baa2:0x3ffd10e0
0x400894f0: invoke_abort at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/esp32/./panic.c:553

0x400895ef: abort at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/esp32/./panic.c:553

0x400d4b83: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40082623: heap_caps_free at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/heap/./heap_caps.c:274 (discriminator 1)

0x40082a65: _free_r at /home/oliv/Documents/Programmation/ESP32/toolchain/esp-idf/components/newlib/./syscalls.c:42

0x4012b976: btdm_rom_ref_init at ??:?

0x4012ba1c: r_rw_pre_main at arch_main.c:?

0x4012baa2: btdm_controller_task at ??:?
```

chandan2002x, 2017-11-17:
Same issue here.
Compiling in Cygwin and esp-idf-v2.1
Any BT program is running. Interestingly, the A2DP_Sink was running initially. After I update the git repo, the ESP is restarting with heap_memory erros as mentioned below.
And all other BT program also behaving the same. Compilation passed, but memory issue at starting time.
 

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
                   
configsip: 0, SPIWP:0xee
                                                      
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
       
mode:DIO, clock div:2
                                                         
load:0x3fff0018,len:4
                                                         
load:0x3fff001c,len:5720
                                                      
load:0x40078000,len:0
                                                         
ho 12 tail 0 room 4
                                                           
load:0x40078000,len:13760
                                                     
entry 0x40079020
                                                              
[0;32mI (73) boot: Detected ESP32[0m
                                        
[0;32mI (33) boot: ESP-IDF v3.0-dev-806-gde750e99-dirty 2nd stage bootloader[0
m
                                                                             
[0;32mI (33) boot: compile time 20:26:27[0m
                                 
[0;32mI (35) boot: Enabling RNG early entropy source...[0m
                  
[0;32mI (40) boot: SPI Speed      : 40MHz[0m
                                
[0;32mI (44) boot: SPI Mode       : DIO[0m
                                  
[0;32mI (48) boot: SPI Flash Size : 4MB[0m
                                  
[0;32mI (52) boot: Partition Table:[0m
                                      
[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[
0m
                                                                            
[0;32mI (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[0m
                                                                          
[0;32mI (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[0m
                                                                          
[0;32mI (78) boot:  2 factory          factory app      00 00 00010000 00200000
[0m
                                                                          
[0;32mI (86) boot: End of partition table[0m
                                
[0;32mI (90) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3e6
b4 (255668) map[0m
                                                           
[0;32mI (187) esp_image: segment 1: paddr=0x0004e6dc vaddr=0x3ffc0000 size=0x01
934 (  6452) load[0m
                                                         
[0;32mI (190) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0x8e
848 (583752) map[0m
                                                          
[0;32mI (397) esp_image: segment 3: paddr=0x000de868 vaddr=0x3ffc1934 size=0x01
10c (  4364) load[0m
                                                         
[0;32mI (399) esp_image: segment 4: paddr=0x000df97c vaddr=0x40080000 size=0x00
[... 39 lines cut ...]
[0;32mI (192) main: RAM left: 173480[0m
                                     
[0;32mI (252) main: hardware initialized[0m
                                 
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed
: file "C:/msys32/esp-idf-v2.1/components/heap/heap_caps.c", line 274, function:
 heap_caps_free
                                                               
abort() was called at PC 0x400d44c3 on core 0
                                 
                                               
                               
Backtrace: 0x4008aa50:0x3ffd65d0 0x4008ab4f:0x3ffd65f0 0x400d44c3:0x3ffd6610 0x4
00826ab:0x3ffd6640 0x40082c09:0x3ffd6660 0x4000bec7:0x3ffd6680 0x4014f676:0x3ffd
66a0 0x4014f71c:0x3ffd66d0 0x4014f7a2:0x3ffd66f0
                              
                                                  
                            
Rebooting...
                                                                  
ets Jun  8 2016 00:22:57
                                                      
                          
                                                    
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

Oliv4945, 2017-11-25:
Hi,
Solved on my side, I updated submodules...

projectgus, 2017-12-01:
Thanks @Oliv4945 

@Cooler1989 @chandan2002x can you confirm you have current submodules as well? ("git status" will indicate, also some warnings should be logged when you run "make")

Cooler1989, 2017-12-05:
I did checkout latest version of master and updated submodules (git submodule update). It seems everything is normal now. I didn't see frequent restart and any warnings about assert traps. I tried few bluetooth examples which previously ended in assert causing reset by watchdog.

Yulong-espressif, 2017-12-15:
@Cooler1989 So have you fix your problem now?

Cooler1989, 2017-12-15:
I was just getting stared with your project and running few examples just to see how it works. It didn't work so I posted issue. Now these few examples seems to work correctly - no assert trap hit. Summing up I think it fixes my problem.

TianHao-Espressif, 2017-12-18:
@Cooler1989 , it's just the submodule not update, right?

Cooler1989, 2017-12-18:
I should close the issue? Sorry, didn't get that. From my site everything is ok. I am closing issue. Thanks guys. Sorry for delay - it was my first issue.

closed 2017-12-18 by Cooler1989