espressif/esp-idf #1251: Precise time of data package sending

liulingliuf, 2017-11-10:
hi,
Recently, I handle a project using ESP32. I desire to get the precise time that a data package sent by WIFI.
How should I get this?
thanks

liuzfesp, 2017-11-26:
Hi @liulingliuf currently we have no API for this, maybe we can support it in IDF release 3.1.

jack0c, 2018-01-14:
@liulingliuf Do you mean the packets send by ESP32? If LWIP provides the packets, we need to see where to store the time. Also,the time will not be 100% accurate.

liuzfesp, 2018-03-05:
@@liulingliuf I'm now considering to support this feature in IDFv3.1, need you help to provide more detailed requirements about this feature. Do you want to get the send time for each packet that sent by LWIP or by esp_wifi_80211_tx?

closed 2018-04-04 by FayeY