espressif/esp-idf #1253: Bluetooth Frequency / Channel Hopping

bmetman, 2017-11-10:
Dear Espressif, 

I had a question regarding the Bluetooth libraries, specifically the BLE api. Quoting [Wikipedia](https://en.wikipedia.org/wiki/Bluetooth#Implementation), Bluetooth supports 40 channels in BLE, using 2 MHz spacing:

> Bluetooth operates at frequencies between 2402 and 2480 MHz, or 2400 and 2483.5 MHz including guard bands 2 MHz wide at the bottom end and 3.5 MHz wide at the top This is in the globally unlicensed (but not unregulated) industrial, scientific and medical (ISM) 2.4 GHz short-range radio frequency band. Bluetooth uses a radio technology called frequency-hopping spread spectrum. Bluetooth divides transmitted data into packets, and transmits each packet on one of 79 designated Bluetooth channels. Each channel has a bandwidth of 1 MHz. It usually performs 800 hops per second, with Adaptive Frequency-Hopping (AFH) enabled.Bluetooth low energy uses 2 MHz spacing, which accommodates 40 channels.

Is it possible to set a connection to a specific channel / frequency using any of the Bluetooth api calls? I haven't found any such calls looking through the api docs. From what I can measure the ESP32 BLE library automatically switches through channels. I'd like to set connections to one channel. 

Thank you for your help.

negativekelvin, 2017-11-10:
Frequency hopping is part of Bluetooth.  Without it you wouldn't be doing Bluetooth.

bmetman, 2017-11-13:
Ok, then I'll assume it is not possible. Thank you.

closed 2017-11-13 by bmetman