espressif/esp-idf #1254: [feature request]: Get the current WiFi protocol mode (b/g/n)

farizadp, 2017-11-10:
Hello,

Is it possible to add functions to get the current associated WiFi protocol (b/g/n)? for both AP and ST mode?
I looked at the wifi stack debug messages that it prints out the WiFi protocol, but could not find this in the source code.

I (25462) wifi: n:1 0, o:1 0, ap:1 1, sta:0 0, prof:1
I (25462) wifi: station: 84:3a:4b:08:2c:d8 join, AID=1, **n,** 20
I (50202) wifi: station: 84:3a:4b:08:2c:d8 leave, AID = 1
I (50202) wifi: n:1 0, o:1 0, ap:1 1, sta:0 0, prof:1
I (116442) wifi: n:1 0, o:1 0, ap:1 1, sta:0 0, prof:1
I (116442) wifi: station: 84:3a:4b:08:2c:d8 join, AID=1, **b,** 20
I (138902) wifi: station: 84:3a:4b:08:2c:d8 leave, AID = 1
I (138902) wifi: n:1 0, o:1 0, ap:1 1, sta:0 0, prof:1

Regards,
Dion

farizadp, 2017-11-21:
@FayeY , just out of my curiosity. You added the TW hashtag in the title. What does the TW# stands for?

FayeY, 2017-11-22:
Hi, it is an issue ID in our internal issue tracking system.

zhangyanjiaoesp, 2018-01-04:
@farizadp  We will add this feature as soon as possible.

farizadp, 2018-01-09:
@zhangyanjiaoesp Thanks !! Do you know an ETA for this?

zyj1993, 2018-01-12:
may be 1~2 months

zhangyanjiaoesp, 2018-01-12:
may be 1~2 months

farizadp, 2018-02-16:
Hello,

Is there anyone who can give any update on this feature request?

Thanks
Dion

zhangyanjiaoesp, 2018-03-15:
@farizadp I'm sorry we have so many things to do, so this feature was delayed. We will finish this in on week.

farizadp, 2018-04-05:
Hello.

Any update regarding this ?

Thanks,
Dion

zhangyanjiaoesp, 2018-04-09:
@farizadp the code has been finished, but we have something else to be merged together, so it still need some time.

FayeY, 2018-05-09:
Has merged. Please have a try with the latest IDF.

closed 2018-05-09 by FayeY