espressif/esp-idf #1256: Ability to send raw IEEE802.11 data frames

seppestas, 2017-11-10:
I need a way to send raw 4-address IEEE 802.11 data frames (Type 0b10 subtype 0b0000). The old , non-public [`ieee80211_freedom_output` function](https://github.com/Jeija/esp32free80211/blob/master/components/free80211/free80211.c) worked fine for this, but the new [`esp_wifi_80211_tx` function](https://github.com/espressif/esp-idf/blob/master/docs/api-guides/wifi.rst#wi-fi-80211-packet-send) seems to [overwrite the frame counter of the frames](https://github.com/espressif/esp-idf/issues/677#issuecomment-343216096), even if en_sys_seq is set to false, and it does not properly increment the frame counter.

I understand [the main reason for not wanting to make a public raw wifi API is that it could be used for sending things like DEAUTH messages](https://esp32.com/viewtopic.php?f=13&t=586&p=2648&hilit=wifi_send_pkt_freedom#p2648). An API that only allows to send valid IEE802.11 frames and limits the allowed frame types (e.g no management frames) would solve this issue and greatly increase user-friendliness.

Please also make it possible to set the frame counter or at least make the frame counter increment correctly.

liuzfesp, 2017-11-26:
This is already in our feature TODO list, but not in a very high priority it will be support in IDF3.1

liuzfesp, 2017-11-30:
@seppestas until not I also not sure when to release IDF3.1, now we are trying to release IDF3.0. For IDF3.1 maybe  it will be released in 1st month of 2018.

seppestas, 2018-01-22:
Nearing the end of the first month of 2018 ^^. Is there any update on this?

liuzfesp, 2018-01-22:
@seppestas still no update yet because too much work to do, but we will try to met v3.1 schedule.

closed 2018-05-18 by igrr