espressif/esp-idf #1257: Calling tcpip_adapter_get_dns_info too soon before setting changing the IP of an AP causes DHCPS to issue IPs from the "old" IP range.

MrSurly, 2017-11-10:
Short version: calling `tcpip_adapter_get_dns_info` too soon before setting changing the IP of an AP causes DHCPS to issue IPs from the "old" IP range.

The code below is for the ESP32 port of MicroPython.  It allows you to pass 4 parameters to either get or set the "ifconfig" of an interface.  The parameters are IP, netmask, gateway, and DNS server.

Recently the code was updated so that setting the DNS for an access point (AP) would work correctly.  
In doing so, a call to `tcpip_adapter_get_dns_info` was added to get the current DNS info, and a call to `tcpip_adapter_set_dns_info` was added to set the DNS info.

However, it seems that if you call `tcpip_adapter_get_dns_info` ("get DNS") and then quickly call `tcpip_adapter_set_ip_info` ("set IP"), then the  DHCPS server will give out address from the _old_ IP of the device.

Removing `tcpip_adapter_get_dns_info` entirely, or **adding a short delay** after calling it fixes this issue.

Per the code below, removing the line marked ### AAAA (`tcpip_adapter_get_dns_info`) fixes the DHCPS problem. OR adding the line marked #### BBBB (`usleep(100000);`) fixes the problem, but removing line #### CCCC `tcpip_adapter_set_dns_info` _does not_ fix the issue.

```C
STATIC mp_obj_t esp_ifconfig(size_t n_args, const mp_obj_t *args) {
    wlan_if_obj_t *self = MP_OBJ_TO_PTR(args[0]);
    tcpip_adapter_ip_info_t info;
    tcpip_adapter_dns_info_t dns_info;
    tcpip_adapter_get_ip_info(self->if_id, &info);
    tcpip_adapter_get_dns_info(self->if_id, TCPIP_ADAPTER_DNS_MAIN, &dns_info); ### AAAA
    usleep(100000); ### BBBB
    if (n_args == 1) {
        // get
        mp_obj_t tuple[4] = {
            netutils_format_ipv4_addr((uint8_t*)&info.ip, NETUTILS_BIG),
            netutils_format_ipv4_addr((uint8_t*)&info.netmask, NETUTILS_BIG),
            netutils_format_ipv4_addr((uint8_t*)&info.gw, NETUTILS_BIG),
            netutils_format_ipv4_addr((uint8_t*)&dns_info.ip, NETUTILS_BIG),
        };
        return mp_obj_new_tuple(4, tuple);
    } else {
        // set
        mp_obj_t *items;
        mp_obj_get_array_fixed_n(args[1], 4, &items);
        netutils_parse_ipv4_addr(items[0], (void*)&info.ip, NETUTILS_BIG);
        if (mp_obj_is_integer(items[1])) {
            // allow numeric netmask, i.e.:
            // 24 -> 255.255.255.0
            // 16 -> 255.255.0.0
            // etc...
            uint32_t* m = (uint32_t*)&info.netmask;
            *m = htonl(0xffffffff << (32 - mp_obj_get_int(items[1])));
        } else {
            netutils_parse_ipv4_addr(items[1], (void*)&info.netmask, NETUTILS_BIG);
        }
        netutils_parse_ipv4_addr(items[2], (void*)&info.gw, NETUTILS_BIG);
        netutils_parse_ipv4_addr(items[3], (void*)&dns_info.ip, NETUTILS_BIG);
        // To set a static IP we have to disable DHCP first
        if (self->if_id == WIFI_IF_STA) {
            esp_err_t e = tcpip_adapter_dhcpc_stop(WIFI_IF_STA);
            if (e != ESP_OK && e != ESP_ERR_TCPIP_ADAPTER_DHCP_ALREADY_STOPPED) _esp_exceptions(e);
            ESP_EXCEPTIONS(tcpip_adapter_set_ip_info(WIFI_IF_STA, &info));
            ESP_EXCEPTIONS(tcpip_adapter_set_dns_info(WIFI_IF_STA, TCPIP_ADAPTER_DNS_MAIN, &dns_info));
        } else if (self->if_id == WIFI_IF_AP) {
            esp_err_t e = tcpip_adapter_dhcps_stop(WIFI_IF_AP);
            if (e != ESP_OK && e != ESP_ERR_TCPIP_ADAPTER_DHCP_ALREADY_STOPPED) _esp_exceptions(e);
            ESP_EXCEPTIONS(tcpip_adapter_set_ip_info(WIFI_IF_AP, &info));
            ESP_EXCEPTIONS(tcpip_adapter_set_dns_info(WIFI_IF_AP, TCPIP_ADAPTER_DNS_MAIN, &dns_info)); #### CCCC
            ESP_EXCEPTIONS(tcpip_adapter_dhcps_start(WIFI_IF_AP));
        }
        return mp_const_none;
    }
}
```

liuzfesp, 2017-11-26:
@MrSurly we will check it next weekly and update you.

zhangyanjiaoesp, 2017-11-28:
@MrSurly 
I'm sorry , I have test as you said, but I can't reproduce your problem. Can you post your result here?
By the way, I have something to suggest:
(1) In AP mode, you just set the static IP address, but you didn't set the lease IP address, which need to call 

> tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_SET,TCPIP_ADAPTER_REQUESTED_IP_ADDRESS,&lease, sizeof(lease));

(2) In AP mode, if you want to set DNS Server, you need to enable it firstly,  that means to call 

> tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_SET,TCPIP_ADAPTER_DOMAIN_NAME_SERVER,&dns_enable, sizeof(dns_enable));  dns_enable = 1

MrSurly, 2017-11-30:
@zhangyanjiaoesp 

This is strange.  The AP DHCPS code has been working fine until the addition of `tcpip_adapter_get_dns_info`.

Looking at the code, calling `tcpip_adapter_dhcps_start` has the same effect as `tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_SET,TCPIP_ADAPTER_REQUESTED_IP_ADDRESS,&lease, sizeof(lease));`

I'll try to create a re-producible test program.

zhangyanjiaoesp, 2017-12-04:
@MrSurly 
Can you exclude the impact of the other parts of your program on this code? I still can't reproduce your problem. It will be better if you can provide a test program .

MrSurly, 2017-12-06:
@zhangyanjiaoesp I can't seem to reproduce this outside of MicroPython -- I'm not sure why.  I'm closing this until I can gather more information.  Thanks for your effort.

closed 2017-12-06 by MrSurly