espressif/esp-idf #1259: Porting mxml library

Pratikhyadav, 2017-11-11:
Hello,

I want to port mxml library in esp-idf where i want to use my static libmxml.a. For that i added One folder in Component , it's structure is as following:

-- component
-- libxml
-- include
(contain header file)
-- lib
(contain lib file libmxml.a)
--component.mk
COMPONENT_ADD_LDFLAGS=$(COMPONENT_PATH)/lib/libmxml.a -l$(COMPONENT_NAME)
include $(IDF_PATH)/make/component_common.mk

--main
--main.c
contain application code
-- component.mk
include $(IDF_PATH)/make/component_common.mk

after compiling this project i get following error:

/esp-idf-v2.1/components/libmxml/lib/libmxml.a: error adding symbols: File format not recognized

Am i Doing anything wrong?
Can anyone provide suggestion regarding this or guidance to port library in esp-idf?

igrr, 2017-12-08:
Based on the error, libmxml.a was seen by the linker, but it is not a valid library file.

You can check whether the library file is valid by running the following commands (from the directory where libmxml.a is stored):

1. Check the format of archive file itself:
```
$ file libmxml.a
```
Should output:
```
libmxml.a: current ar archive
```

2. Check the format of files contained inside the archive file:
```
xtensa-esp32-elf-ar -t libmxml.a | xargs -I % sh -c "xtensa-esp32-elf-ar -x libmxml.a %; file %; rm %;"
```
This should output one line for each file in the archive, and all the lines should look like
```
some_file_name.o: ELF 32-bit LSB relocatable, Tensilica Xtensa, version 1 (SYSV), not stripped
```

igrr, 2018-01-08:
Please feel free to reopen if this is still an issue.

closed 2018-01-08 by igrr