espressif/esp-idf #1260: FreeRTOS xTaskList should be extended to show xCoreID for SMP

dmcnaugh, 2017-11-11:
Since ESP-IDF has added SMP features to FreeRTOS it would be very helpful to see what core each task is running on.

I have made this change with a quick and dirty mod and it has helped me debug some task conflict problems very effectively as I had wrongly assumed all tasks were on the PRO core unless I had pinned them to the APP core, but this was not the case.

chanibal, 2017-11-13:
@dmcnaugh, maybe post the dirty mod here, so it can be a starting point?

dmcnaugh, 2017-11-18:
OK, here's a patch file for the quick and dirty mod I made.

[Issue#1260.patch.zip](https://github.com/espressif/esp-idf/files/1484329/Issue.1260.patch.zip)

dmcnaugh, 2018-06-14:
I have create a Pull-Request #2064 Added: display xCoreID in vTaskList that has a proposed resolution to this issuie