espressif/esp-idf #1264: Buffer overflow in ipc.c

drwheeler, 2017-11-13:
In components/esp32/ipc.c
``80    const char* task_names[2] = {"ipc0", "ipc1"};
 81     for (int i = 0; i < portNUM_PROCESSORS; ++i) {
 82         s_ipc_sem[i] = xSemaphoreCreateBinary();
 83         portBASE_TYPE res = xTaskCreatePinnedToCore(ipc_task, task_names[i], CONFIG_IPC_TASK_STACK_SIZE, (void*) i,
 84                                                     configMAX_PRIORITIES - 1, &s_ipc_tasks[i], i);
 85         assert(res == pdTRUE);
 86     `}`

If portNUM_PROCESSORS is ever defined to be more than 2, task_names will be accessed past its end and tasks will have memory contents as names.

liuzfesp, 2017-11-26:
Thanks @drwheeler point it out, it's safe currently because we only support 2 CPU. Maybe it will be better if we can change it to support any number of CPUs.

zhangyanjiaoesp, 2017-11-27:
Thanks @drwheeler , I have modify this code

zhangyanjiaoesp, 2017-11-28:
Thanks @drwheeler, I have modify this code.  
commit ID:https://gitlab.espressif.cn:6688/idf/esp-idf/merge_requests/1605

closed 2017-11-28 by FayeY