espressif/esp-idf #1265: BLE Scanning doesn't stop after interval

nkolban, 2017-11-13:
This one is rather tricky to describe ... so I may be a little wordy.

I have a BLE library written in C++.  Part of the function performs a scan looking for advertised external (to ESP32) BLE devices.  I eventually call the API called `esp_ble_gap_start_scanning()` passing in a duration in seconds for which I wish to scan.  I start receiving advertisements and all is well.  After the duration interval passes, I receive the event that indicates that scanning has stopped.  Great.  All is perfect.

However ... things go wrong in one strange area.  If I use a Samsung cell phone and load the "nRFConnect" application and ask that application to start advertising as a BLE server, what we find is that after starting the scanning on the ESP32, we start to see received advertisements but ... after the duration at which scanning should stop .. it simply doesn't.  The ESP32 simply continues to receive advertisements and we see no indication of the end event that indicates that scanning has completed.

I can recreate my issue at will.  If I set my Samsung/Android phone to NOT advertise, after the scanning internal period, my app stops (correctly) with the event indicating that the period has elapsed.  If I start my Samsung/Android phone to perform advertising, the scanning does not end.

I am unable to find any issues with me ESP32 ESP-IDF coding and hence am opening an issue against the ESP-IDF BLE component itself.  Maybe others will recognize the puzzle.  The original report can be found in the following issue:

https://github.com/nkolban/esp32-snippets/issues/190

mringwal, 2017-11-15:
Hi @nkolban. If you like, you can try BTstack's gap_le_advertisement example and use a timer there to stop after some time to see if it's a bug in the BLE part of the Bluetooth Controller or in the Bluedroid stack. Knowing where to look could make debugging easier.

TianHao-Espressif, 2017-11-15:
@nkolban , we have use another hw timer to stead of freertos timer to  fix the timer start/stop fail bug in e24ea6207ba09aa073449482eecc354203788799. If your version is more previous, please try this version.

nkolban, 2017-11-18:
Howdy my friend,
I sat down today to attempt to report more on the issue.  The first thing I did was to download the very latest ESP-IDF and, at the same time, I also replaced my sdkconfig with the latest version available from the template.  I re-built my BLE application and ran again ... and now I can no longer recreate the issue.  I'm going to assume that either I was back-level on my ESP-IDF release or else I had *something wrong* in my sdkconfig.  Either way, I am good and working now so will close this issue.  MANY thanks for taking the time to respond.

closed 2017-11-18 by nkolban