espressif/esp-idf #1266: BLE, is supported muti-connect (from several clients) for ESP32 in peripheral/server/slave state?

ayavilevich, 2017-11-13:
Hi,

As I have read, Bluetooth 4.1 spec allows a peripheral to have several connections at the same time. Prior to 4.1 a peripheral could only have one connection. AFAIK ESP32 is a 4.2 device.
I am trying to implement this with ESP32 and it almost works but some things break. Not sure if the blame is on the server or clients yet. Before going deeper I would like to get a confirmation whether this is known to be supported on the ESP32 BT stack. Can somebody confirm if this already works or if this is something that is scheduled for the future?

mringwal, 2017-11-15:
Hi. The ESP32 Bluetooth Controller can handle multiple peripheral roles. I've tested this with our [BTstack port](https://github.com/bluekitchen/btstack/tree/master/port/esp32).

I don't know how well it this is supported by the Bluedroid stack.

In general, being in more than one connections isn't trivial and you have to expect a "worse" connection than with a single one. If you're not too aggressive with the connection interval and don't need to stream at full speed, it should work in most cases.

ayavilevich, 2017-11-20:
@FayeY why was this closed without an explanation or a response?

chegewara, 2017-11-20:
Maybe this is the answer?
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/gattc_multi_connect

ayavilevich, 2017-11-20:
@chegewara, this is similar but not the same. The example you linked is for a multi-connect client. I am inquiring about the same for a server role.

ayavilevich, 2017-11-20:
@FayeY, no problem. Thanks for checking that out. Can you clarify if you are working on an example or if the implementation is incomplete in the current framework? I guess what I am asking is, should I wait for a future release or can I code my own example already?

Cheers to the team.

xiewenxiang, 2017-12-08:
Hi @ayavilevich 

If you want the server connect to more than one client, You just need start ADV after the previous connection was established.

ayavilevich, 2017-12-08:
@xiewenxiang , this is what you would expect. I tried it, but it doesn't work well. See https://github.com/espressif/esp-idf/issues/1263

FayeY, 2018-01-05:
Hi, multi-connection of BLE is supported in the latest IDF. So I'm going to close this issue. If there is any bug reporting of the multi-connection, please submit a new issue. Thanks.

closed 2018-01-05 by FayeY

ayavilevich, 2018-02-11:
Multi-connection might be working for gatt client or in some cases for gatt server but it is not working well. 
I have documented a specific use case in #1263.

The latest reproduce instructions are at https://github.com/espressif/esp-idf/issues/1263#issuecomment-364751036 . If everybody who contributed to this discussion would try to reproduce and post their results, perhaps this will help find a common cause.

NateZimmer, 2019-10-13:
Has anyone had any luck with this? My experience is:
- Advertisement stops as soon as 1 client connects to ESP32 GATT server 
- Advertisement resumes after client disconnects

chegewara, 2019-10-13:
> Has anyone had any luck with this? My experience is:
> 
> * Advertisement stops as soon as 1 client connects to ESP32 GATT server
> * Advertisement resumes after client disconnects

There is plenty of reports about multi-connections, especially in arduino. What you have to do is to re-start advertising when central is connected.

NateZimmer, 2019-10-13:
Thanks, the resume of advertisement appears to be working which is what I needed. Need another phone to test the multi-connect case. 

FYI 1 line added here: 
```c
    case ESP_GATTS_CONNECT_EVT: {
        esp_ble_conn_update_params_t conn_params = {0};
        memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
        /* For the IOS system, please reference the apple official documents about the ble connection parameters restrictions. */
        conn_params.latency = 0;
        conn_params.max_int = 0x20;    // max_int = 0x20*1.25ms = 40ms
        conn_params.min_int = 0x10;    // min_int = 0x10*1.25ms = 20ms
        conn_params.timeout = 400;    // timeout = 400*10ms = 4000ms
        ESP_LOGI(GATTS_TAG, "ESP_GATTS_CONNECT_EVT, conn_id %d, remote %02x:%02x:%02x:%02x:%02x:%02x:",
                 param->connect.conn_id,
                 param->connect.remote_bda[0], param->connect.remote_bda[1], param->connect.remote_bda[2],
                 param->connect.remote_bda[3], param->connect.remote_bda[4], param->connect.remote_bda[5]);
        gl_profile_tab[PROFILE_A_APP_ID].conn_id = param->connect.conn_id;
        //start sent the update connection parameters to the peer device.
        esp_ble_gap_update_conn_params(&conn_params);
+++     esp_ble_gap_start_advertising(&adv_params); // Added this line for multiple connects?
        break;
```

Might be handy to add this in the example itself

GabrielapessoaS, 2019-10-28:
does anyone have a example of how to connect more than 2 Esp32 at the same time with BLE?