espressif/esp-idf #1268: esp32 pretrans incompatible with full-duplex

jstar88, 2017-11-13:
I need pretrans time delay in full-duplex SPI. 
Is there a fix?

```
    //The hardware looks like it would support this, but actually setting cs_ena_pretrans when transferring in full
    //duplex mode does absolutely nothing on the ESP32.
    SPI_CHECK(dev_config->cs_ena_pretrans==0 || (dev_config->flags & SPI_DEVICE_HALFDUPLEX), "cs pretrans delay incompatible with full-duplex", ESP_ERR_INVALID_ARG);
```

ginkgm, 2017-11-14:
No, due to silicon issue, there'll be no fix for this for some time.

Please use half-duplex mode, or try to handle the CS line yourself:

1. assign no CS line (-1) to the driver
2. add pre_trans_cb and post_trans_cb to the driver
3. control the gpio in the callback.

we're considering adding this into the driver, though it's obviously not a hardware solution.

jstar88, 2017-11-14:
@ginkgm 
That's a great idea, however is not permitted to use post_trans_cb in full-duplex.
Any suggestions? thanks

ginkgm, 2017-11-15:
@jstar88 I mean the `pre_cb` and `post_cb` in the `spi_device_interface_config_t` struct. see `void lcd_spi_pre_transfer_callback(spi_transaction_t *t) ` in the spi_master example. iirc the pre and post cb can be used whether using full-duplex or not.

sorry for the inaccurate name of member.

like:
```
    spi_device_interface_config_t devcfg={
        .clock_speed_hz=10*1000*1000,               //Clock out at 10 MHz
        .mode=0,                                //SPI mode 0
        .spics_io_num=PIN_NUM_CS,               //CS pin
        .queue_size=7,                          //We want to be able to queue 7 transactions at a time
        .pre_cb=cs_active,  //Specify pre-transfer callback to handle CS line
        .post_cb=cs_incative,
    };

void cs_active(spi_transaction* trans)
{
  gpio_set_level(PIN_NUM_CS, 0);
}

void cs_inactive(spi_transaction* trans)
{
  gpio_set_level(PIN_NUM_CS, 1);
}
```

jstar88, 2017-11-17:
@ginkgm 
thanks,that's mine:

`struct timeval tv_start, tv_stop;
void lcd_spi_pre_transfer_callback(spi_transaction_t *t)
{
	gpio_set_level(QTOUCHSPI_SS, 0);
	gettimeofday(&tv_start, NULL);
	gettimeofday(&tv_stop, NULL);
	while(tv_stop.tv_usec - tv_start.tv_usec < 2)
	{
		gettimeofday(&tv_stop, NULL);
	}
	//vTaskDelay(1 / portTICK_RATE_MS);
}

void lcd_spi_post_transfer_callback(spi_transaction_t *t)
{
	//vTaskDelay(1 / portTICK_RATE_MS);
	gettimeofday(&tv_start, NULL);
	gettimeofday(&tv_stop, NULL);
	while(tv_stop.tv_usec - tv_start.tv_usec < 1)
	{
		gettimeofday(&tv_stop, NULL);
	}
	gpio_set_level(QTOUCHSPI_SS, 1);
}`

ginkgm, 2017-11-17:
Great. And you may know that vTaskDelay may cause task switching, which is not good in a pre/post cb called in the ISR. 

There're also 3 APIs (or method) for you to do this: (see `components/log/log.c`)
1. `xTaskGetTickCount()` , 1 tick for 10ms. e.g.
```
t_ms = xTaskGetTickCount() * (1000 / configTICK_RATE_HZ); //configTICK_RATE_HZ is often 100.
```
2. `uint32_t esp_log_early_timestamp()`, return in ms. 

3. 
``` 
extern uint32_t g_ticks_per_us_pro; //this is pre-defined and written somewhere in the bootloader, don't change this!!!!!!!!
t_us = xthal_get_ccount() / g_ticks_per_us_pro;
```

jstar88, 2017-11-17:
@ginkgm 
thanks, I need to wait just 1us in pre_cb and 20ns in post.
I could wait 1ms too but it's a waste of time

jstar88, 2017-12-10:
@ginkgm 
an issue just showed up.
My code, with gettimeofday() function, work fine.
However if there is another task calling gettimeofday() function like:
 
```
struct timeval tv_startx, tv_stopx;
void anotherTask(Display* obj)
{
     gettimeofday(&tv_startx, NULL);
     //... stuff
     gettimeofday(&tv_stopx, NULL);
     printf("cttime = %li[us]\n",tv_stopx.tv_usec-tv_startx.tv_usec);
}
```

then the esp32 crash

```
abort() was called at PC 0x40082a03 on core 1
0x40082a03: lock_acquire_generic at /home/nicola/esp/esp-idf/components/newlib/./locks.c:146

Backtrace: 0x40088aa0:0x3ffc0ee0 0x40088b6b:0x3ffc0f00 0x40082a03:0x3ffc0f20 0x40082af1:0x3ffc0f50 0x400e19d2:0x3ffc0f70 0x40082886:0x3ffc0f90 0x400d8060:0x3ffc0fb0 0x400d31c1:0x3ffc0fd0 0x40083c3b:0x3ffc0ff0 0x400821c9:0x3ffc1020 0x400d1de3:0x00000000
0x40088aa0: invoke_abort at /home/nicola/esp/esp-idf/components/esp32/./panic.c:572

0x40088b6b: abort at /home/nicola/esp/esp-idf/components/esp32/./panic.c:572

0x40082a03: lock_acquire_generic at /home/nicola/esp/esp-idf/components/newlib/./locks.c:146

0x40082af1: _lock_acquire at /home/nicola/esp/esp-idf/components/newlib/./locks.c:165

0x400e19d2: get_boot_time at /home/nicola/esp/esp-idf/components/newlib/./time.c:103

0x40082886: _gettimeofday_r at /home/nicola/esp/esp-idf/components/newlib/./time.c:187

0x400d8060: gettimeofday at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/syscalls/../../../.././newlib/libc/syscalls/sysgettod.c:12

0x400d31c1: lcd_spi_post_transfer_callback at /home/nicola/esp/esp32_project/main/touch/QtouchSPI.c:221

0x40083c3b: spi_intr at /home/nicola/esp/esp-idf/components/driver/./spi_master.c:318

0x400821c9: _xt_lowint1 at /home/nicola/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

0x400d1de3: esp_vApplicationIdleHook at /home/nicola/esp/esp-idf/components/esp32/./freertos_hooks.c:85
```
To avoid the crash, i must remove gettimeofday() from one of the two task.
Seems like a sort of ISR - thread safe concurrency

ginkgm, 2017-12-22:
@jstar88 as you can see, the crash happens in the ISR. Functions used in the ISR should be chosen carefully.

Similar to other libc functions, read, write, printf, and so on — all can't be used in ISR.

We suggest you to use ``esp_timer_get_time`` instead.

FayeY, 2018-01-11:
Hi @jstar88 , did the above suggestion help you resolve the issue?

closed 2018-01-26 by FayeY

jstar88, 2018-03-05:
no, I had to remove the "time delay" code.
Luckly, the spi slave work fine anyway