espressif/esp-idf #1269: (s)printf positional parameters do not work correctly

akomakom, 2017-11-13:
From peeking at newlib.h, positional parameters are supposed to be supported (unless I need to activate that somehow?). 

**newlib.h**:
```
/* Positional argument support in printf functions enabled.  */
/* #undef _WANT_IO_POS_ARGS */
```

Yet I am seeing some really odd results, especially with floats:

    printf("printf positional test: %1$d\n",  3);
    printf("printf positional test: %1$f\n",  0.3f);
    printf("printf normal test: %d\n",  3);
    printf("printf normal test: %f\n",  0.3f);

Produces

```
printf positional test: 3
printf positional test: 15380079272638060709447077316139445090788965024504223438471168.000000
printf normal test: 3
printf normal test: 0.300000
```

Also, when using multiple positional parameters, anything **starting with float parameter is broken**:

`printf("printf positional test: %1$d ; %2$f ; %3$d\n",  3, 0.3f, 4);`

> printf positional test: 3 ; 0.000000 ; 1070805811

even if I **switch the order**, the first mention of float is still breaking things:

`printf("printf positional test2: %1$d ; %3$f ; %2$d\n",  3, 4, 0.3f);`

> printf positional test2: 3 ; 1.681847 ; 4

Finally, if I don't use float formats, it works correctly:

`printf("printf positional test3: %1$d ; %3$d ; %2$d\n",  3, 4, 5);`

> printf positional test3: 3 ; 5 ; 4

### Environment
I'm on esp-idf hash 8e04b8615331583840312c140818e8ca5aa5ea2a
Ubuntu 17.10

My includes are as follows:

```
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include "lwip/netdb.h"
#include "lwip/dns.h"

#include "mbedtls/platform.h"
#include "mbedtls/net_sockets.h"
#include "mbedtls/esp_debug.h"
#include "mbedtls/ssl.h"
#include "mbedtls/entropy.h"
#include "mbedtls/ctr_drbg.h"
#include "mbedtls/error.h"
#include "mbedtls/certs.h"
```

igrr, 2018-01-08:
Thanks for submitting this issue @akomakom, it was a very interesting one to investigate.
I have traced this issue to a [14-year-old change in newlib](https://github.com/bminor/newlib/commit/759a097e72f42d8c834b9de64affe1975e591930).

There is a piece of code in `get_arg` function (which implements retrieval of arguments from va_list in specified order) which scans for a beginning of format sequence. Original implementation was (for platforms which define MB_CAPABLE) is the correct one:

https://github.com/bminor/newlib/blob/9588ff7555057ea1f62fda35e60428b5fa2ce943/newlib/libc/stdio/vfprintf.c#L2086-L2091

Notice that it increments `fmt` and then checks if the last character was `%`. So once `%` is found, `fmt` points past it.

Implementation for !MB_CAPABLE does not work the same way:

https://github.com/bminor/newlib/blob/9588ff7555057ea1f62fda35e60428b5fa2ce943/newlib/libc/stdio/vfprintf.c#L2096-L2097

It does not increment `fmt` after `%` is found, so `fmt` still points at `%`, which causes the subsequent state machine to immediately bail out. The code defaults to loading the next "int" argument, which is why positional formats seems to work in some cases (when arguments are 32-bit and when argument numbers are sequential).

Apparently all "big" newlib users which have _WANT_IO_POS_ARGS are also MB_CAPABLE, which is why the issue has been in the code for such a long time.

We'll be preparing patches for this and fixing this in IDF.

igrr, 2019-04-12:
Preview release of the toolchain which fixes this issue is now available, see https://www.esp32.com/viewtopic.php?f=10&t=7400&p=31257#p41667.

igrr, 2019-06-20:
Sorry, forgot to update this issue. The toolchain has been updated in https://github.com/espressif/esp-idf/commit/ca323fbaad540cf66cb267c0f749ad24f3a9b211 and includes newlib version with the fix.

closed 2019-06-20 by igrr