espressif/esp-idf #1271: BLE multiple operations in parallel

dimonomid, 2017-11-14:
When I try to fire, for example, a read of some charateristic, and then before it completes, a write to another characteristic, I get this:

```
E (8087) BT: bta_gattc_enqueue: already has a pending command!!
```

And my callback isn't called at all. So, the fact that the callback isn't called definitely looks like a bug. Other than that, a question: is it intended not to be able to execute parallel request?

Thanks.

mringwal, 2017-11-15:
GATT/ATT does not support concurrent requests. I think you may send a write_without_response at any time but you cannot do multiple reads or writes in parallel. Does the gatt write function has an error code? I'm not familiar with Bluedroid. On BTstack, you would get an error for the gatt write function and no callback either.

Cicy-Chen, 2017-11-20:
Which version of BLE idf do you use，V2.1 or V3.0？

Yulong-espressif, 2017-12-18:
@dimonomid We have added a command queue in the Host stack now, means that you can send more then one command at the same time. Please update the master.

dimonomid, 2017-12-20:
@Yulong-espressif thanks a lot!

closed 2017-12-21 by FayeY

ravdeepboparai, 2020-04-26:
Is this command queue in Host stack also implemented for the AT engine of ESP for all the commands?

xiewenxiang, 2020-10-13:
ESP32-AT is developed based on ESP-IDF, So it behaves like Bluedroid.