espressif/esp-idf #1273: UDP and TCP server not accepting connections after WDT reboot.

CharlesSeartech, 2017-11-14:
Hi,

I currently use the ESP-WROOM-32 for a project and we use two client TCP connections, two server TCP connections and a UDP server connection.
If the board is powered-up or hard reset from monitor the ESP32 boots correctly and all the connections work as expected, but if the WDT reboots the board the TCP and/or UDP servers don't accept connections. 
UDP setup:
udp_server_socket = socket(AF_INET, SOCK_DGRAM, 0);
struct sockaddr_in server_addr;
server_addr->sin_family = AF_INET;
server_addr->sin_port = htons(EXAMPLE_DEFAULT_PORT);
server_addr->sin_addr.s_addr = htonl(INADDR_ANY);
bind(udp_server_socket , (struct sockaddr *)&server_addr, sizeof(server_addr));
sockaddr_t remote_addr;
uint32_t  socklen = sizeof(remote_addr);
length = recvfrom(udp_server_socket , CtrlRxUdpBuffer, CONFIG_TCP_MSS, 0, &remote_addr, &socklen);

All the setup calls are checked for errors.

chanibal, 2017-11-14:
What IDF version are you using?

Have you tried setting the second argument of `listen(tcp_server_socket, 1);` to something higher than one?

CharlesSeartech, 2017-11-14:
Hi Chanibal,

Using IDF 2.1.

Backlog set to 1 as I only want to allow one connection at a time.

chanibal, 2017-11-14:
I meant that as for debugging purposes only.

CharlesSeartech, 2017-11-16:
After further testing on the UDP I can supply the following info:

CharlesSeartech, 2017-11-17:
I have found what causes the UDP to not receive broadcast messages, can someone please let me know how to solve it now?

Working boot:
I (550) MyWiFi: WiFi wifi_init.
I (551) wifi: wifi firmware version: 7bd4899
I (551) wifi: config NVS flash: enabled
I (553) wifi: config nano formating: disabled
I (567) wifi: Init dynamic tx buffer num: 16
I (568) wifi: Init data frame dynamic rx buffer num: 8
I (568) wifi: Init management frame dynamic rx buffer num: 8
I (571) wifi: wifi driver task: 3ffce6c8, prio:23, stack:4096
I (576) wifi: Init static rx buffer num: 8
I (580) wifi: Init dynamic rx buffer num: 8
I (584) wifi: Init rx ampdu len mblock:7
I (588) wifi: Init lldesc rx ampdu entry mblock:4
I (592) wifi: wifi power manager task: 0x3ffd3024 prio: 21 stack: 2560
I (598) MyWiFi: WiFi set_storage.
I (603) MyWiFi: esp_wifi_stop.
I (1007) MyWiFi: wifi_init_sta mode.
I (1007) MyWiFi: wifi_init_sta config.
I (1008) MyWiFi: wifi_init_sta start.
I (1039) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0
I (1040) wifi: mode : sta (a8:f7:25:a4:ae:30)
I (1041) MyWiFi: wifi_init_sta finished.
I (1051) MyWiFi: Waiting for connection to AP.
I (2005) wifi: n:8 0, o:1 0, ap:255 255, sta:8 0, prof:1
I (2995) wifi: state: init -> auth (b0)
I (2999) wifi: state: auth -> assoc (0)

**I (3003) wifi: state: assoc -> run (10)
I (3020) wifi: connected with MY_Wifi, channel 8**

Not working boot:
I (549) MyWiFi: WiFi wifi_init.
I (550) wifi: wifi firmware version: 7bd4899
I (550) wifi: config NVS flash: enabled
I (552) wifi: config nano formating: disabled
I (567) wifi: Init dynamic tx buffer num: 16
I (568) wifi: Init data frame dynamic rx buffer num: 8
I (568) wifi: Init management frame dynamic rx buffer num: 8
I (571) wifi: wifi driver task: 3ffce6c8, prio:23, stack:4096
I (576) wifi: Init static rx buffer num: 8
I (580) wifi: Init dynamic rx buffer num: 8
I (584) wifi: Init rx ampdu len mblock:7
I (588) wifi: Init lldesc rx ampdu entry mblock:4
I (592) wifi: wifi power manager task: 0x3ffd3024 prio: 21 stack: 2560
I (598) MyWiFi: WiFi set_storage.
I (603) MyWiFi: esp_wifi_stop.
I (1007) MyWiFi: wifi_init_sta mode.
I (1007) MyWiFi: wifi_init_sta config.
I (1008) MyWiFi: wifi_init_sta start.
I (1054) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0
I (1056) wifi: mode : sta (a8:f7:25:a4:ae:30)
I (1056) MyWiFi: wifi_init_sta finished.

I (1067) MyWiFi: Waiting for connection to AP.
I (2021) wifi: n:8 0, o:1 0, ap:255 255, sta:8 0, prof:1
I (3010) wifi: state: init -> auth (b0)
I (3031) wifi: state: auth -> assoc (0)

**I (3034) wifi: state: assoc -> run (10)
I (3089) wpa: GTK is already in use in entry 2, it may be an attack, ignor it.
I (3090) wifi: connected with MY_Wifi, channel 8**

The section's in bold show the only difference between the working and not working boot. I found that this is the cause as I enabled the IP_DEBUG and could see that the IP stack does not receive any messages not specifically addressed to the ESP32, i.e. the IP stack does not play out the following messages on the not working boot:
ip4_input: UDP packet to DHCP client port 1900
ip4_input: packet not for us.

I can supply the IP_DEBUG messages if required.

Regards,
Charles

liuzfesp, 2017-11-26:
@CharlesSeartech I think this issue should already be fixed. Please @TimXia help to check it.

liuzfesp, 2017-11-26:
@CharlesSeartech could you help to provide the commit ID of IDF? You can paste the log of reboot to here.

TimXia, 2017-11-27:
@CharlesSeartech This is a already known issue during IDF development. Please update IDF to the latest version IDF v2.1.1 to fix this issue.

CharlesSeartech, 2017-11-27:
Thanks guys, V2.1.1 did solve the problem.

closed 2017-11-27 by CharlesSeartech