espressif/esp-idf #1275: Bluetooth Controller: Link Key not emitted on re-pairing

mringwal, 2017-11-15:
The ESP32 Bluetooth Controller fails to emit a link key when the reconnecting and the remote has deleted its link key. Btw. this might be related to #1082 - more incorrect behaviour related to pairing/encryption.

Steps to reproduce
- app creates an outgoing classic connection
- as it has a stored link key, it requests authentication and provides the link key upon request
- the remote device has deleted the link key for some reason and does not provide a link key
- as there's no shared link key, pairing starts and succeed
However, the Controller doesn't emit the new link key. In the next re-connect, both sides have a different link key (ESP32 the outdated one and the remote the current one) and the connection fails. It's crucial that the link key is emitted and this affect any Bluetooth stack running on the ESP32.

Here's the full log as 
[esp32-re-pair.txt (txt)](https://github.com/espressif/esp-idf/files/1474227/esp32-re-pair.txt) and as 
[esp32-re-pair.pklg.zip (.pklg for Wireshark)](https://github.com/espressif/esp-idf/files/1474230/esp32-re-pair.pklg.zip)

It can be reproduced e.g. using a patched version of spp_streamer_client of BTstack. You can change the BD Addr of the remote device in btstack_main and also the RFCOMM Channel.

[spp_streamer_client.c.zip](https://github.com/espressif/esp-idf/files/1474249/spp_streamer_client.c.zip)

mringwal, 2017-12-22:
Hello Espressiv. Any feedback on this bug in the Bluetooth Controller? 

With this one and maybe #1082 fixed, I guess the ESP32 could be recommended for production use with Bluetooth Classic as long as Wifi isn't used.

mywang-espressif, 2018-03-21:
Hi @mringwal , thanks for the detailed description and useful analysis. I can understand the scenario and I have reproduced the issue with your example. Next I will look into the logic and try to find out the cause.

mringwal, 2018-03-21:
Great news! Let me know when you have a version to test

mywang-espressif, 2018-03-22:
Hello, I have found out the cause and attached is the library file for your test. Please let me know if it fixes this issue.
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/1837319/libbtdm_app.a.zip)

mywang-espressif, 2018-03-22:
BTW, this library works with current ESP-IDF master branch, commit 0a83733d and you don't need to patch other files.

mringwal, 2018-03-22:
excellent. I'll give it a try later and report back

mringwal, 2018-03-22:
hi @mywang-espressif I'm happy to confirm that the link key is emitted for repairing in the provided library. Thanks!

I'd leave this issue open until the fix is integrated on the esp32-bt-lib and available in esp-idf master.

mywang-espressif, 2018-03-23:
Glad to hear that. I will further verify the change and put forward a merge request. I will let you know after it is integrated.

mywang-espressif, 2018-04-02:
Thanks @igrr. hi @mringwal , the patched library file is merged in to master branch. You can have a try.

closed 2018-04-04 by FayeY