espressif/esp-idf #1276: Bluetooth Controller: ACL Connections survive reboot via Reset button

mringwal, 2017-11-15:
If there's an active Classic connection and I press the Reset button on my ESP32 Thing, it does a full boot up. Asides from the MCU rebooting, BTstack also sends a HCI Reset Command to reset the Bluetooth Controller.

Still, I get the error trying to reconnect to the device where the connection was open before:
[HCI EVENT]  Connection Complete - ACL Connection Already Exists - F4:0F:24:3B:1B:E1

Of course, this should not happen. Even if the MCU reset doesn't affect the Bluetooth Controller, the HCI Reset must reset it's internal state.

Logs as .txt and .pklg as usual: 
[esp32-connection-survives-reset.zip](https://github.com/espressif/esp-idf/files/1474280/esp32-connection-survives-reset.zip)

TianHao-Espressif, 2018-04-03:
@mringwal, HCI Reset Command will reset all the software and hardware state, but it not cause send LMP_detach. And press the reset button on esp32 means reset the hardware without any software de-initialise, so remotes device don't know your board is leaving until supervision time.

Maybe we can evaluate add LMP_detach when do hci reset.

mringwal, 2018-04-03:
@TianHao-Espressif Let me explain my reasoning, maybe I'm missing something (and I don't know much about Bluetooth Classic Link Layer).

On start (either by HCI Reset or power up), I would expect that the controller initializes everything, so there are no physical connections. 
How does the controller know that there is/was a connection? How can it decide to reject the connection request?

TianHao-Espressif, 2018-09-03:
ringwal . On latest master branch. I tested it with our tool(not bt stack), and there's no the same problem. 

I have tried our bluedroid and another host tool over UART HCI. Both of the method are working that ESP32 could reconnect to the remotes devices after reset.

So could you test it based on latest master branch? Thanks.

mringwal, 2018-09-25:
@TianHao-Espressif thanks for checking this. I did a few tests with the current version and did not see the issue anymore. Looks good now.

closed 2018-09-25 by mringwal