espressif/esp-idf #1277: esp_bt_controller_mem_release doesn't work

rojer, 2017-11-15:
In my test APP, i have BT compiled in but it is not started and not initialized, so `esp_bt_controller_mem_release` should release all the memory. But it doesn't.
`ESP_ERROR_CHECK( heap_caps_add_region(mem_start, mem_end));` crashes, and if i remove the check, the two calls look like this:
```
I (1415) BTDM_INIT: Release DRAM [0x3ffae6e0] - [0x3ffaff10]
I (1425) BTDM_INIT: Release DRAM [0x3ffb0000] - [0x3ffc0000]
```
both are unsuccessful and return `-1` (`ESP_FAIL`).
`[0x3ffae6e0] - [0x3ffaff10]` does not match any region in `soc_memory_regions`.
heap map at boot looks like this:
```
[Nov 15 18:32:56.607] I (800) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[Nov 15 18:32:56.607] I (806) heap_init: At 3FFD03E0 len 0000FC20 (63 KiB): DRAM
[Nov 15 18:32:56.629] I (813) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
[Nov 15 18:32:56.629] I (819) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[Nov 15 18:32:56.629] I (825) heap_init: At 400937A4 len 0000C85C (50 KiB): IRAM
```

negativekelvin, 2017-11-16:
https://github.com/espressif/esp-idf/blob/e6afe28bafe5db5ab79fae213f2e8e1ccd9f937c/components/soc/esp32/soc_memory_layout.c#L77

chegewara, 2017-11-16:
Actually it does work, but with small issue. When i call 	`esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT); ` from app_main, its very first line in app_main, then it works without issues. When i call it later, didnt check yet what place it is, but for sure before BT init, then i have heap corrupt issues.

TianHao-Espressif, 2017-11-23:
@negativekelvin , the range is valid. The end address is also align to 4bytes. The problem is fixed in 2d148cd34b01aedf66a4fc846de0c296b790ee5a

closed 2017-12-11 by FayeY