espressif/esp-idf #1280: Broken ULP ADC measurements during wakeup

JcBernack, 2017-11-16:
I'm using a ULP program to constantly take measurements from the ADC. This program is even running when the system is not in deep-sleep to not lose any data. Problem is, that the ADC gives bogus values when the system is waking up from deep-sleep. It seems like the values are always the maximum possible readings, i.e. 4095 for 12-bit measurements.

With a sample rate of 10Hz there are one or two bad samples, with 40Hz there are 7 oder 8, so there is a window of about 200ms during wakeup where measurements are broken. Disabling wifi cuts the time in half.

So my questions are:
- Did you know about this issue? Is this expected to happen?
- And is there something I can do about it, other than filtering out values with a value of 4095?

krzychb, 2017-11-17:
Hi @JcBernack,
Could you post a minimum code that exhibits this issue?

JcBernack, 2017-11-17:
Hi @krzychb,
there you go https://github.com/JcBernack/esp-ulp-adc-bug

krzychb, 2017-11-17:
Wow! Very nice and right to the point example!

I see similar issue. 

For your latest application compiled with the current master release from the GitHub the result is as follows:
```
I (99) main: samples taken 506
I (99) main: first bad sample at 499
I (99) main: number of bad samples 6
I (99) main: resetting counters
I (99) main: entering deep sleep
```
If the bootloader messages are enabled, then wakeup cycle takes longer, and the extra samples take are broken as well:

```
I (195) main: samples taken 515
I (195) main: first bad sample at 499
I (195) main: number of bad samples 14
I (195) main: resetting counters
I (195) main: entering deep sleep
```

I will contact the hardware designer working with analog part to check what can be done about this issue.

Thank you for reporting it and providing the test code.

igrr, 2017-11-18:
This period of 4095 values likely corresponds to the time when we reconfigure SAR ADC to seed the random number generator (https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/bootloader_start.c#L478, https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/bootloader_start.c#L543). Could you try commenting out these two lines and repeat the test?

krzychb, 2017-11-18:
Hi @igrr,

Yes! Reconfiguration of the SAR ADC to seed the random number generator was  the cause.

```
I (196) main: samples taken 514
I (196) main: first bad sample at 514
I (196) main: number of bad samples 0
I (196) main: resetting counters
I (196) main: entering deep sleep
 ```
Thank you for quick support and pinning it down in the first attempt :+1:

krzychb, 2017-11-19:
I am wondering if we need to seed the random number generator on each soft reboot.
Would it be enough to seed it only on power on boot?

igrr, 2017-11-19:
Random number generator state is not preserved when the system goes into deep sleep, so we do need to seed it. Since (afaik) there is no way to seed PRNG from software, we can't just save the seed value in RTC memory and write it into PRNG upon reset.

JcBernack, 2017-11-20:
Thanks for your quick responses and confirming the issue!
Sounds like there isn't gonna be an easy fix for this. I guess I have to skip samples with the max value for now.

krzychb, 2017-11-21:
>  I guess I have to skip samples with the max value for now.

Did you consider checking one of registers set by [bootloader_random_enable](https://github.com/espressif/esp-idf/blob/15a61459610d6ead47d0a940bba65215c37a541d/components/bootloader_support/src/bootloader_random.c#L63) and then cleared by [bootloader_random_disable](https://github.com/espressif/esp-idf/blob/15a61459610d6ead47d0a940bba65215c37a541d/components/bootloader_support/src/bootloader_random.c#L115) to decide when to skip doing measurements? This would be a more generic approach to avoid potentially skipping "legitimate" max samples, when the actual measurement is indeed at the maximum, not because of seeding the random number generator.

A separate question is what particular register to check, that is not altered elsewhere when the main CPUs are up and running  :smile:

JcBernack, 2017-11-22:
> A separate question is what particular register to check

That is where I stopped considering that option 😄

negativekelvin, 2017-11-23:
Maybe RTC_CNTL_ENT_RTC

nevercast, 2017-11-28:
SENS_SAR2_EN_TEST is also possible I'd guess.

closed 2017-12-07 by igrr