espressif/esp-idf #1283: Reported ADC voltage using esp-idf 2.0 incorrect or wrongly documented

seppestas, 2017-11-17:
I'm trying to measure the voltage of a battery using the ADC of the ESP32 (ESP-WROOM-32) with ESP-IDF 2.0.
The maximum voltage I'm trying to measure is 2.1V (Lipo battery with a 1/2 voltage divider). Since the ESP-IDF documentation states the following:
> When VDD_A is 3.3V:
>
> 0dB attenuaton (ADC_ATTEN_0db) gives full-scale voltage 1.1V
> 2.5dB attenuation (ADC_ATTEN_2_5db) gives full-scale voltage 1.5V
> 6dB attenuation (ADC_ATTEN_6db) gives full-scale voltage 2.2V
> 11dB attenuation (ADC_ATTEN_11db) gives full-scale voltage 3.9V (see note below)

I'm using a 6dB attenuation (`ADC_ATTEN_6db`). I'm measuring on pin 34 (ADC1 channel 6) using 12-bit sampling (`ADC_WIDTH_12Bit`).

Relevant source code snippet:
```C
adc1_config_width(ADC_WIDTH_12Bit);
adc1_config_channel_atten(ADC1_CHANNEL_6, ADC_ATTEN_11db); // Measure up to 2.2V
int battery_voltage = adc1_get_voltage(ADC1_CHANNEL_6);
if (battery_voltage < 0) {
    printf("Failed to read battery voltage\n");
} else {
    printf("Battery voltage: %X -> %fV\n", battery_voltage, 2.2*(float)(battery_voltage)/0xFFF);
```

The voltage reported by `adc1_get_voltage` seems to be incorrect. When measuring a voltage of 1.750V the following is returned: Battery voltage: EFB -> 2.060317V.

I also tried assuming the full scale voltage is 1.6V (3.3V power supply divided by 2). This gives a closer result of 1.484347V, but it is still pretty far away from the expected 1.750V.

The ADC already returns a full-scale voltage at around 1.85V.

Is this a problem with the ADC, ESP-IDF or the documentation?

negativekelvin, 2017-11-17:
Calibration and linearization of ADC was only introduced in master 3.0-dev branch.

seppestas, 2017-11-18:
Does that mean measuring a voltage using the ADC as described in the docs
is pretty much impossible? It woulde be nice if at least the docs mentioned
this.

Since I’m using functions that only seem to work on esp-idf v2.0 I’m kind
of stuck with it. Any recomendations on how to calibrate the ADC manualy?

chegewara, 2017-11-18:
This is most relevant line:

> 11dB attenuation (ADC_ATTEN_11db) gives full-scale voltage 3.9V (see note below)

As you can read the full-scale is for 3.9V then if you have esp32 powered with 3.3V then you need to "rescale" ADC this way:
measured voltage = measured value* 3.9V/3.3V

Now you have:
1.75 * 3.9/3.3 = 2.068 (battery real voltage)

OtherCrashOverride, 2017-11-19:
Based on experience with real hardware, even with esp-idf 3.x, the ADC readings are extremely unreliable.  This is due to ADC reference voltages that vary wildly from module to module.

The API docs suggest outputting the internal reference voltage to a GPIO and measuring it.  Then this value should be used as a calibration value.  However, from practical experience, even this measured value will not be accurate.

The eventual workaround I ended up using is to use a known good reference voltage and to "reverse calculate" the real reference voltage from it.  Since the equation for calibration used in esp-idf has not been made known, I ended up "brute forcing" the calculation.  The end result using this method was +/- 0.01V within the measured LiPo battery voltage.

JcBernack, 2017-11-20:
@OtherCrashOverride Can you share a code snippet of your method?

seppestas, 2017-11-21:
> The eventual workaround I ended up using is to use a known good reference voltage and to "reverse calculate" the real reference voltage from it. Since the equation for calibration used in esp-idf [...]

I’m assuming this works when only using a single module, however I’m working on a project with multiple ESP32 modules (and later maybe the chips directly), so a more scalable solution would be nice.

chegewara, 2017-11-21:
I dont know how real its equation in this vid, but maybe its worth to check(at 5:35):
https://www.youtube.com/watch?v=RlKMJknsNpo

PS there is a lot wrong informations about read values

Dazza0, 2021-09-23:
ADC on the ESP32 (and later chips) requires calibration. Please refer to the [ADC calibration documentation](IDFGH-128) for more details. Closing this issue for now. Please feel free to reopen.

closed 2021-09-23 by Dazza0